realsenseai/librealsense #4097: D435 error when plugged in at PC boot

piraog, 2019-05-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.4 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.21.0                          | 
| Language                            |  opencv/python                          | 
| Segment			|  others                           | 

### Issue Description
When opening Realsense Viewer or trying to run a python script to stream from the camera, i get the following error:
[image]

Now, the camera usually works perfectly fine, and I'm really happy with the performances. But from time to time I get stuck with this error and forced to replug the device.
From what I read on the now closed issue #2932 and my experience, it seems like it happens when I restart my computer with the camera plugged in and don't replug the camera afterwards.

Following your instructions on #2932 I've logged the infos when I try to open the viewer (copied for the fw-logger):
[fwlog.log](https://github.com/IntelRealSense/librealsense/files/3231101/fwlog.log)

I've also tried to update the firmware to 5.11.6.200, according to your past advices, but couldn't find this version of the firmware on the driver page... It only goes up to 5.11.4.

The problem is that in my use case, my computer operates in a remote place and must restart now and then, so I often end up stuck with this error and replugging the device isn't a long term solution for me.

Do you have any solution since #2932 ? Is it being looked into?
Is their any way to prevent being stuck in this situation, maybe a protocol to follow before shutting down?

Thanks in advance for your response.

AntoineWefit, 2019-05-29:
We actually encounter a similar issue which is preventing the cameras to work properly after any boot.
A reliable fix is indeed to unplug/replug, but it's quite inconvenient...
Updating the firmware also doesn't do anything (for now), plus it's quite a pain to update our codes to match the new firmware/libraries and hope that this problem is solved.

I notice that your camera is plugged in as USB 2.1, is that a consequence of the problem or intentional ? Maybe scheduling a **USB reset** at your computer boot could do the trick ? Haven't tested it yet though...

> I've also tried to update the firmware to 5.11.6.200, according to your past advices, but couldn't find this version of the firmware on the driver page... It only goes up to 5.11.4.

Not sure if it's going to help you but here is a copy : [Signed_Image_UVC_5_11_6_200.zip](https://github.com/IntelRealSense/librealsense/files/3231334/Signed_Image_UVC_5_11_6_200.zip)

piraog, 2019-05-29:
Thanks for the input!

The device is indeed plugged in as 2.0, it's intentional as I don't need the 3.0 tranfer rates, 2.0 is enough for my usage.
I'll try to update the firmware with your driver and do a USB reset to see if it could be used as a temporary fix.

Will keep you updated

piraog, 2019-05-29:
Ok,
I tried upgrading the firware to the one you gave me but wasn't able to do so, the camera kept disapearing from the updating tool when I tried to upload the new version.
As for restarting the USB, I'm not sure it would solve the problem, I'll try when I get the oportunity (I'm running other tasks from the same computer and curently can't shut it down.

AntoineWefit, 2019-05-29:
> I tried upgrading the firware to the one you gave me but wasn't able to do so, the camera kept disapearing from the updating tool when I tried to upload the new version.

I've never experienced this issue, so this is kind of a mystery for me...
Anyway, I'm warning you about the fact that the 5.11.6.200 version of the firmware has been chronologically released **before** the 5.11.4 one, and as you noticed, isn't available anymore. I don't know much about what's going on with it, but I wouldn't rely on it to develop something stable...
It appeared that I kept a copy of it on my computer, so use it at your risk if it's deprecated :)

> As for restarting the USB, I'm not sure it would solve the problem, I'll try when I get the oportunity (I'm running other tasks from the same computer and curently can't shut it down.

Please let me how it goes for you !

piraog, 2019-05-31:
Small update on the matter, 
I tried reproducing the problem with the same camera on an almost identical setup (but different computer), and didn't encounter the same issue, with both USB 2.0 or 3.0 connections.

m9dfukc, 2019-06-01:
hi,
I encounter a similar issue. I have three D435 devices (FW 05.11.06.200) connected to a usb 3.2 hub. One camera repeatably disappears after reboot, if I unplug and replug the camera it works again. Unfortunately this is not an option in my scenario. A solution to this issue is much needed!

piraog, 2019-06-05:
Any insight from the intel realsense dev team?

AntoineWefit, 2019-06-06:
@piraog @m9dfukc we also had communication problems with our D400 cameras (not sure if it's related to your issues) and we tried to trigger a **USB reset** before every utilization : didn't work...
BUT using the `hardware_reset` function and waiting a bit after instead seemed to do the trick, we will do further tests though...

Hope it'll help !

antoniodourado, 2019-06-13:
I'm facing the very same issue here.  
Manually unplug/replug isn't a option for me either. I need this issue solved.
Hope dev team help us :(

@AntoineWefit which `hardware_reset` function are you talking about?

antoniodourado, 2019-06-14:
I'm also adding `lsusb` when my system is up but realsense isn't detected.
```bash
antonio@mintpc:~$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 046d:c32b Logitech, Inc. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 005: ID 2109:0813 VIA Labs, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 0d8c:0014 C-Media Electronics, Inc. 
Bus 001 Device 005: ID 2109:2813 VIA Labs, Inc. 
Bus 001 Device 003: ID 046d:c332 Logitech, Inc. 
Bus 001 Device 002: ID 046d:0a5b Logitech, Inc. 
Bus 001 Device 008: ID 1b1c:0c09 Corsair 
Bus 001 Device 006: ID 051d:c812 American Power Conversion 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
antonio@mintpc:~$ rs-enumerate-devices 
No device detected. Is it plugged in?
```

And when I plug it in:
```bash
Bus 002 Device 010: ID 8086:0b3a Intel Corp.
```

AntoineWefit, 2019-06-14:
@antoniodourado you can find it [here](https://github.com/IntelRealSense/librealsense/blob/79dda7f00ebde9564f1a2053b8ab95b7ba41ac69/include/librealsense2/hpp/rs_device.hpp#L87-L96).
Call it like so : `device.hardware_reset()`; also, waiting a bit after seems to help (in our case).
Hope this'll help !

m9dfukc, 2019-06-14:
@AntoineWefit @antoniodourado  hardware reset doesn't help if your device is not detected in the first place?!

AntoineWefit, 2019-06-14:
@m9dfukc as I said, this solution seems to be a fix for **our case** (we still need to do furthers tests though).
As you said, if your device is not detected, I highly doubt that this'll work sadly...

But if the device is incorrectly detected (connected as USB 2.X when it should be USB 3.X for example), `hardware_reset` actually help quite a bit. In the original post, it looks like the camera is indeed detected as it appears in the Viewer.

Again, we clearly doesn't have the exact same issue, nor the same setup; I'm just sharing a fix hoping it'll help somebody else :)

piraog, 2019-06-14:
@AntoineWefit @m9dfukc As you can see in the picture from my original post, the device is recognised and detected, so a hardware reset should be possible. I still have not got the time to fully try this though. And it is not a long term solution.

@antoniodourado I'm not sure we have the exact same issue

m9dfukc, 2019-06-14:
@piraog yes you are right this seems to be a different issue.

antoniodourado, 2019-06-14:
@piraog @m9dfukc @AntoineWefit yeah, probably a different issue :( I'm going to post a new topic then. Thank you anyway!

closed 2019-07-09 by RealSenseCustomerSupport