realsenseai/librealsense #4098: IR dots visible in RGB camera

ShettyHarapanahalli, 2019-05-29:
| | |
|---|---|
|**librealsense**|2.22.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435|
|**Serial Number**|745412070314|
|**Firmware Version**|05.10.03.00|
|**Advanced Mode**|YES|
|**Usb Type Descriptor**|3.2|

I see the IR dots visible in the RGB camera, what parameters do I adjust to fix it. 

[image]

MartyG-RealSense, 2019-05-29:
There was a similar case recently at the Intel Support site's RealSense forum.  I recommended using a 'Visual Preset' configuration called 'Remove IR Pattern' to remove the pattern without affecting the image as significantly as turning off the IR Projector component that generates the pattern.

https://forums.intel.com/s/question/0D50P00004IyLdpSAF/d435-pink-ir-dots-in-rgb-image

AntoineWefit, 2019-05-29:
I'm also interested !
But is this preset working for the D435 though @MartyG-RealSense ?
Just asking after seeing [this](https://github.com/IntelRealSense/librealsense/issues/1827#issuecomment-394559279).

MartyG-RealSense, 2019-05-29:
Oh, the Remove IR Pattern preset is D415 only.  I do apologize.  

Turning the IR projector off may be the only quick fix then, and it is one that worked for the D435 user in that Intel Support case.  I find that typically, this can have a noticable effect on the illumination level of an IR pattern, though it does not seem to affect the RGB image much.

dorodnic, 2019-06-03:
Hi @ShettyHarapanahalli @AntoineWefit 
Instead of disabling laser completely, you could try reducing laser power. From my experience, these dots are only visible in RGB at short range, where laser emission is significant.

closed 2019-07-01 by RealSenseCustomerSupport