realsenseai/librealsense #4104: can not install realsense for R200

RZz96, 2019-05-29:
Hi
i can not install pyrealsense using  setup.py from https://github.com/toinsson/pyrealsense

and i get the error:
LINK : fatal error LNK1181: cannot open input file 'realsense.lib'
error: command '\\AppData\\Local\\Programs\\Common\\Microsoft\\Visual C++ for Python\\9.0\\VC\\Bin\\amd64\\link.exe' failed with exit status 1181

MartyG-RealSense, 2019-05-29:
Have you installed the older Legacy Librealsense, please?    The installation instructions link on the Pyrealsense site goes to an instructions page for the current RealSense SDK 2.0, which does not have support for the R200.  Legacy Librealsense can be found at the link below:

https://github.com/IntelRealSense/librealsense/tree/v1.12.1

RZz96, 2019-05-30:
Thank you for reply .
i have built the library . how can i install it for python ?

MartyG-RealSense, 2019-05-30:
Are you able to do a Python installation with PyPy, please?

https://pypi.org/project/pyrealsense/

RZz96, 2019-05-30:
yes  . and now in virtual environment i want to install pyrealsense for R200 .

MartyG-RealSense, 2019-05-30:
If you are using a virtual machine, that complicates things a bit, as Librealsense traditionally has some problems handling virtual machines.  The installation documentation says:

"Due to the USB 3.0 translation layer between native hardware and virtual machine, the librealsense team does not support installation in a VM.  If you do choose to try it, we recommend using VMware Workstation Player, and not Oracle VirtualBox for proper emulation of the USB3 controller".

RZz96, 2019-05-30:
thanks for the reply . 
i uninstall the pyrealsense2 on base and tried  to install the pyrealsense for R200 and i got the error : 
LINK : fatal error LNK1181: cannot open input file 'realsense.lib'

MartyG-RealSense, 2019-05-30:
This subject is a bit outside of my area of experience, but I did locate a case where another user who was trying to install Pyrealsense for an R200 got the same error.  Please read downwards from the comment linked to below to see the advice that was given to them.

https://github.com/IntelRealSense/librealsense/issues/3648#issuecomment-479189012

RZz96, 2019-05-30:
no it didnt help

MartyG-RealSense, 2019-05-30:
I do not know what else to suggest, unfortunately.  I see that you have opened a new case for this particular question, which is a good idea.  I would recommend closing this case and please waiting for a response on your new question from an Intel team member on this forum, please (I do not work for Intel).  Good luck!

RZz96, 2019-05-30:
ok thank you

closed 2019-05-30 by RZz96