realsenseai/librealsense #4105: D435 NOT RECOGNIZED AS USB3.1 DEVICE

giovanidiniz, 2019-05-29:
| | |
|---|---|
|**librealsense**|2.22.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense USB2|
|**Serial Number**|841512070830|
|**Firmware Version**|05.11.04.00|
|**Advanced Mode**|YES|
|**Usb Type Descriptor**|2.1|

Intel realsense is not recognized as USB3.1 device and hence does not have all functionality avialable. 
Interesting enough, even with the USB2 functionality available, I still don't have an RGB Camera option in realsense-viewer.

I tried using different cables, several different firmware and connecting/disconnecting in all sorts of ways. My computer is a new System76 running Ubuntu 18 and the ports are working perfectly fine.

bgroisser, 2019-06-03:
It sounds like you've already tried this, but I've found that the USB3.1 cables are not actually reversible; turning the USBC connector 180 degrees solved this problem for me.

giovanidiniz, 2019-06-03:
Thanks guys. 
I ended up buying a $15 USB-C to USB3 cable from Amazon and that did the trick.
Interesting enough, the cable that came with the camera did not work properly. I mean...

Well, closing this now. Thanks for the ideas!

closed 2019-06-03 by giovanidiniz