realsenseai/librealsense #4106: legacy camera support with new kernels (?)

svenevs, 2019-05-29:
> All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

:heart:

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  F200  | 
| Firmware Version                   | 2.60.0.0 (super old...) | 
| Operating System & Version |   Fedora 29  | 
| Kernel Version (Linux Only)    |  4.18.16                                         | 
| Platform                                 | PC |
| SDK Version                            |  legacy / v1.12.1                          | 
| Language                            |  any                          | 
| Segment			|  any                          | 

### Issue Description

Was running some tests on something on a rig that I know for a fact supported this camera before I upgraded operating systems.  It worked on Fedora 25 and I believe Fedora 27, but now I'm on Fedora 29.  With GCC 8.3.1 I was able to compile things after adding two includes in different files (`#include <functional>`, and `#include <sys/sysmacros.h>` (glibc / sys/types.h changes)) in different places.  I was planning on including a full patch if there was interest in a legacy patch release, but I'm starting to hit problems I don't know anything about.

```console
$ ./examples/cpp-enumerate-devices
Device 0 - Intel RealSense F200:                                                                                                                              
 Serial number: 029140057806                                                                                                                                  
 Firmware version: 2.60.0.0                                                                                                                                   
 USB Port ID: 4-2                                                                                                                                             
 Camera info:                                                                                                                                                 
    DEVICE_NAME         :       Intel RealSense F200                                                                                                          
    DEVICE_SERIAL_NUMBER:       029140057806                                                                                                                  
    CAMERA_FIRMWARE_VERSION:    2.60.0.0                                                                                                                      
 Supported options:                                    min        max       step  default                                                                     
    COLOR_BACKLIGHT_COMPENSATION                       : 0    ... 0           0     0
...
many video modes
...
rs.error: TemperatureControlLoop: opcodes do not match
```

It's from here

https://github.com/IntelRealSense/librealsense/blob/7332ecadc057552c178addd577d24a2756f8789a/src/hw-monitor.cpp#L123-L128

where in my specific case `opCodeAsUint32 == 4294967289` and `opCodeXmit == 188`.  For shiggles I allowed those specific values (it didn't appear to be an endianness thing btw), which will ultimately fail on starting the device with `VIDIOC_S_FMT error 22, Invalid argument`.  I don't know if bypassing the TemperatureControlLoop opcodes check would cause this, but the only interesting thing I could really find out is that the device address appears to be changing.  Is that normal?  I don't remember what it did previously...

```console
$ ./examples/cpp-pointcloud 
RealSense error calling rs_start_source(device:0x2308fb0, source:VIDEO):
    VIDIOC_S_FMT error 22, Invalid argument
$ ./examples/cpp-pointcloud 
RealSense error calling rs_start_source(device:0x18f93c0, source:VIDEO):
    VIDIOC_S_FMT error 22, Invalid argument
$  ./examples/cpp-pointcloud 
RealSense error calling rs_start_source(device:0x72efb0, source:VIDEO):
    VIDIOC_S_FMT error 22, Invalid argument
$  ./examples/cpp-pointcloud 
RealSense error calling rs_start_source(device:0x22c7fb0, source:VIDEO):
    VIDIOC_S_FMT error 22, Invalid argument
$  ./examples/cpp-pointcloud 
RealSense error calling rs_start_source(device:0x1824620, source:VIDEO):
    VIDIOC_S_FMT error 22, Invalid argument
```

https://github.com/IntelRealSense/librealsense/issues/3143#issuecomment-468121126 makes it sound as though downgrading the kernel was the solution.  https://github.com/IntelRealSense/librealsense/issues/105#issuecomment-210006513 mentions BIOS twiddling.  If I had to guess, since I can `gst-launch-1.0 -v v4l2src device=/dev/video0 ! xvimagesink` and view the feed, it's some kind of incompatibility between the kernel version and librealsense v1.12.1.  It'd be cool if I could get this camera working through the API again, it's got some properties that are nice to show off in "benchmarks" because of its range thresholds.

This is a question about a legacy camera on the legacy API, so I totally get it if the answer is #wontfix haha.  But I figured I'd ask the veterans for help since it may not take much to get it working given the information above, and knowledge of how things actually work down in the trenches :slightly_smiling_face:  Thanks for any thoughts!

doisyg, 2019-06-03:
See here: https://github.com/IntelRealSense/librealsense/pull/3929
Currently not working with multiple camera, feel free to contribute

dorodnic, 2019-06-03:
Hi @svenevs 
We are aware that recent changes to the Kernel broke some of the old librealsense functionality, related to metadata nodes (with current proposed w/a by @doisyg that is likely to be merged)
However, `opcodes do not match` should not be related to this.

svenevs, 2019-06-06:
Sorry for delay in testing, had some out of town visitors this week :slightly_smiling_face:  I'm not sure how all the pieces fit together, but #3929 worked for me on this rig without any changes needed!

Not sure how that works with the `opcodes do not match` stuff but I'm going to close this and consider it a win because it works xD

It's really quite amazing that Intel still entertains questions and possible patches to support legacy cameras, especially this far out into the future.

closed 2019-06-06 by svenevs

doisyg, 2019-06-06:
> It's really quite amazing that Intel still entertains questions and possible patches to support legacy cameras, especially this far out into the future.

@svenevs Thanks for the feedback, I am not working for Intel though, I took care of it cause I have more than 50 R200 used on different robots and we need to maintain them