realsenseai/librealsense #4107: Raw image capture from left and right cameras from D435i using Matlab

neocubist, 2019-05-29:
Hi,
I am trying to capture raw image from left and right cameras (stereo, not the rgb) on a D435i. I have very little programming experience and hope to accomplish this with Matlab. I downloaded the SDK with matlab and python options selected, reviewed the list of m files. I did not see an option for raw image capture from each camera. Could anyone point me or advice on how I can accomplish this?

Camera Model - D435i
Firmware Version - 05.10.13.00
OS - Win10
Platform - PC
SDK Version 2.21.0
Language preference - Matlab (first choice) Python (second choice)

Let me know if any more info is needed. I believe this specific question has not been addressed as I spent a considerable amount of time searching before creating a Github account to post this (yes, I am that new :))

Thanks,

lramati, 2019-06-13:
I'm not 100% sure what you mean, but you probably want to look at the config object. Unfortunately there aren't any examples written in Matlab for the class, but the API is the same as in Python so you can easily port the code to Matlab if you prefer it to Python.

As a reference, the config class is used in the early lines of [this Python example](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py)

RealSenseCustomerSupport, 2019-07-05:
----------------------------------------------

@neocubist Any other questions about this ticket? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2019-07-09:
----------------------------------------------

@neocubist If no other questions will close this issue accordingly. Thanks!

closed 2019-07-09 by RealSenseCustomerSupport