realsenseai/librealsense #4108: Incomplete video frame detected on Jetson Nano

EricRohlfs, 2019-05-29:
| | |
|---|---|
|**librealsense**|2.22.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435I|
|**Serial Number**|844212070691|
|**Firmware Version**|05.10.13.00
255.255.255.255|
|**Advanced Mode**|YES|
|**Usb Type Descriptor**|3.2|

Followed the instructions from  
https://github.com/IntelRealSense/librealsense/issues/3759
AND
https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide

When running the realsense viewer, when I turn on RGB, I get hardware notifications "Incomplete video frame detected! Size 65536 out of 1843455 bytes (3%).

When I set the Resolution to 960x540 or lower the errors go away.

kvbogdanov, 2019-05-30:
JetsonHacks channel on Youtube faced such a problem. So they recommend recompiling the kernel. Here is the video https://youtu.be/8ckhQP7Af-g

EricRohlfs, 2019-05-31:
> JetsonHacks channel on Youtube faced such a problem. So they recommend recompiling the kernel. Here is the video https://youtu.be/8ckhQP7Af-g

Thank you, I've been following him too but I was hoping for a bit more commercial support on this especially if I choose to take this to market, I don't want to have to rely solely only on JetsonHacks (I do appreciate his work)

akouri-doordash, 2019-06-03:
I am also having this problem -- but on a standard Ubuntu 18.04 install on an x86 Linux Laptop. Also prefer not to have to recompile kernel.

RealSenseCustomerSupport, 2019-08-23:
----------------------------------------------

The SDK provides official supports for Intel X86/AMD64-based Debian distributions for Ubuntu 16/18.
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

Additionally, the dev team also works with the communities to help to enable RealSense on not officially supported platforms as well as non-LTS Ubuntu kernel. Nevertheless, the SDK and the kernel patches are entirely open-source so the customers are more than welcome to enable and integrate RealSense camera for their specific platforms.

Please let us know if there is any other question. Thank you.

closed 2019-08-30 by RealSenseCustomerSupport