realsenseai/librealsense #4111: D435 Get IR PROBLEM

alexex0723, 2019-05-30:
When I just get Depth stream and Color Stream is working，but when I add IR stream
It show rs2::error 
Does anyone can help me Idk how to deal with

ev-mp, 2019-05-30:
@alexex0723 , can you provide some details ?
- Camera type / OS / SDK version/ Executable
- The error snapshots/logs encountered

alexex0723, 2019-05-30:
Camera Model - D435i
OS - Win10
Platform - PC
SDK Version 2.17.1
My code is like this (I just show some setting part bcz my code is long)
rs2::pipeline pipe;
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
cfg.enable_stream(RS2_STREAM_INFRARED, 640, 480, RS2_FORMAT_Y8, 30);
rs2::pipeline_profile profile = pipe.start(cfg);
rs2::colorizer c; 
auto depth_stream = profile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();
auto color_stream = profile.get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>();
auto infrared_stream = profile.get_stream(RS2_STREAM_INFRARED).as<rs2::video_stream_profile>();

Error is say 0x00007FFC2B299129 (in Depth_camera.exe) An unhandled exception occurred: Microsoft C++ has an exception: rs2::error (memory location 0x000000B5B435C9B0). occur

RealSenseCustomerSupport, 2019-07-02:
----------------------------------------------

@alexex0723 Did you have such IR issue with latest RealSenseViewer(https://github.com/IntelRealSense/librealsense/releases/download/v2.24.0/Intel.RealSense.Viewer.exe)?

RealSenseCustomerSupport, 2019-07-08:
----------------------------------------------

@alexex0723  Any update? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2019-07-15:
----------------------------------------------

@alexex0723  As we didn't get update from your side, will close the issue now. Please feel free to create another new one if you still have issue. Thanks!

closed 2019-07-15 by RealSenseCustomerSupport