realsenseai/librealsense #4112: Error when using D430 camera long time in Win7

ljy345115678, 2019-05-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 05.10.03| 
| Operating System & Version |   Win 7  | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.16.1}                          | 
| Language                            |  {C }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I got the following error when I using D430 camera in win7. 
*******************************************************
receiving interrupt_ep bytes failed: No error
Error receiving message
*******************************************************
The error was occured after my program running in an hour.  I tried many times and simplified my program to simply geting frames from camera, the error would always occur in about one hour.  If I restart the program immediately, the camera works well.  Do anyone knows what's the problem?

RealSenseCustomerSupport, 2019-06-26:
----------------------------------------------

@ljy345115678 Did you try SDK tool RealSenseViewer on your setup to see if still get this issue?

RealSenseCustomerSupport, 2019-07-01:
----------------------------------------------

@ljy345115678 Could you please update? Thanks!

ljy345115678, 2019-07-01:
Yes, I tried. RealSenseViewer doesn't have this problem.

RealSenseCustomerSupport, 2019-07-02:
----------------------------------------------

@ljy345115678 Then it might be your own program issue. Could you please provide your code snippet so that we can look into it? Thanks!

RealSenseCustomerSupport, 2019-07-05:
----------------------------------------------

@ljy345115678 Any other questions about this ticket? Will closet it if no other questions.

closed 2019-07-05 by RealSenseCustomerSupport