realsenseai/librealsense #4113: Perform C++ hardware reset to T265

ineslara21, 2019-05-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {Realsense T265 | 
| Firmware Version                   | 0.0.18.5715 | 
| Operating System & Version |  Ubuntu 16.04  | 
| Kernel Version (Linux Only)    | 4.15.0-50                                        | 
| Platform                                 | x64  |
| SDK Version                            | librealsense 2.22                          | 
| Language                            | C++                         | 
| Segment			|  Robot/others }                          | 

### Issue Description
I would like to know if it is possible to perform hardware reset to T265 tracking camera. 
I was trying to perform it via ROS wrapper, but with the execution of `dev.hardware_reset()` I only get:
  ` An exception has been thrown: hardware_reset is not implemented for this device!`

Then, I came back to the Realsense SDK and I saw the function definition in `device.cpp` file so I understand the exception that has been thrown. 
However, I see some other users talking about the use of hardware reset but I'm not able to perform it. 
Any ideas? I would like to manually reset it on code. 

Thanks in advance.

dorodnic, 2019-05-31:
Hi @ineslara21 
This should be possible, but we need to look up exactly how. I'll update if there's a work-around for now, and we will add it into one of future releases.

indraneelpatil, 2019-07-20:
@dorodnic @ineslara21 Hey, is there any update on this issue? We really need this functionality too.

Thanks in advance!

indraneelpatil, 2019-07-20:
This worked for me : 

`
rs2::device _dev
_dev_sensors = _dev.query_sensors();
_dev_sensors[0].stop();
std::this_thread::sleep_for( std::chrono::seconds( 1 ) ); // wait a little bit
ROS_WARN_STREAM("Waiting:");
std::string module_name= "Tracking Module";
 _dev_sensors[0].start(_sensors_callback[module_name]);
ROS_WARN_STREAM("Hardware Resetting device");
 std::cout<< "Reset called"<<std::endl;
`

RealSenseCustomerSupport, 2019-07-22:
----------------------------------------------

Hi @ineslara21,

You can have a try of below usb reset mechanism and see how it works.
https://github.com/ralight/usb-reset/blob/master/usb-reset.c

ineslara21, 2019-07-23:
Hi @indraneelpatil ,

You're right. It works. It's just what I needed. 
Thanks!!

@RealSenseCustomerSupport thanks for the link, someday I'll try to test how it works.

closed 2019-07-23 by ineslara21

zainmehdi, 2019-12-03:
Has there been any update on hardware reset implementation for T265 ?

radfordi, 2020-01-10:
With the merging of #5213, `hardware_reset` is now supported for the `T265`.  The should be available in `v2.32.0`.

neilyoung, 2020-01-10:
I’m using it for a while already and I confirm  it to work. One has to issue a little timeout (1-2 s) in order to let the cam recover from the hit and proceed with normal op

zainmehdi, 2020-02-19:
Hi @neilyoung. I have been having "Unable to open device interface" when using hardware reset. Can you please share your example ?

zainmehdi, 2020-02-19:
Figures it out. For anyone having the same issue can use this to perform hardware reset

```
rs2::config cfg;
rs2::pipeline pipe;

cfg.resolve(pipe).get_device().hardware_reset();
```

neilyoung, 2020-02-19:
@zainmehdi Sorry, a bit late. I'm doing it a bit different in Python, especially the loop over all devices (overkill) and the pause (necessary)

```
                ctx = rs.context()
                devices = ctx.query_devices()
                for dev in devices:
                    dev.hardware_reset()
                    time.sleep(1)

```

07hokage, 2021-06-16:
> @zainmehdi Sorry, a bit late. I'm doing it a bit different in Python, especially the loop over all devices (overkill) and the pause (necessary)
>
> ```
>                 ctx = [...]

this works well