realsenseai/librealsense #4115: problem to run the realsense example  in c++

RZz96, 2019-05-30:
Hi
i have built the realsense library in c++ and i want to try the example that are included in the legacy.zip. but i got the following error:

rs::error was thrown when calling rs_start_source(device:000002392EF7C530, source:VIDEO):
    no matching media type for  pixel format 59555932

Samvid95, 2019-05-31:
Which camera are you using? Because as far as I know you cannot run legacy examples on D400 series and T200 series cameras.

RZz96, 2019-06-03:
R200

RealSenseCustomerSupport, 2019-06-28:
----------------------------------------------

@RZz96 What example did you try?

RealSenseCustomerSupport, 2019-07-08:
----------------------------------------------

@RZz96 If no other questions about this issue, will close it accordingly. Thanks!

closed 2019-07-08 by RealSenseCustomerSupport