realsenseai/librealsense #4116: Several errors

kevindehecker, 2019-05-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Ubuntu 18.04 | 
| Kernel Version (Linux Only)    | 4.15.0-50-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            | v2.22.0                        | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description

I'm getting a weird set of errors:

Error 1:
```
terminate called after throwing an instance of 'rs2::error'
  what():  Operation failed with error code=1
```
Error contents:		
```
		e	@0x5555559f2190	rs2_error
			args	"\000\000\000\000\000\000\000\0000x5555559f2d50, option:Visual Preset, value:4"	std::__cxx11::string
			exception_type	RS2_EXCEPTION_TYPE_COUNT (8)	rs2_exception_type
			function	"rs2_set_option"	std::__cxx11::string
			message	""	std::__cxx11::string
		h	RS2_EXCEPTION_TYPE_COUNT (8)	rs2_exception_type
```	

Stack:
```
1  __GI_raise               raise.c        51  0x7fffef78be97 
2  __GI_abort               abort.c        79  0x7fffef78d801 
3  ??                                          0x7ffff0180957 
4  ??                                          0x7ffff0186ab6 
5  std::terminate()                            0x7ffff0186af1 
6  __cxa_throw                                 0x7ffff0186d24 
7  rs2::error::handle       rs_types.hpp   140 0x555555577bf4 
8  rs2::options::set_option rs_options.hpp 103 0x5555555799f7 
9  Cam::calib_pose          cam.cpp        678 0x555555572317 
10 Cam::init                cam.cpp        341 0x55555556f728 
11 init                     main.cpp       447 0x5555555c4b9a 
12 main                     main.cpp       568 0x5555555c54e4 
```
At the code
```
    auto range = rs_depth_sensor.get_option_range(RS2_OPTION_VISUAL_PRESET);
    for (auto i = range.min; i < range.max; i += range.step) {
        if (std::string(rs_depth_sensor.get_option_value_description(RS2_OPTION_VISUAL_PRESET, i)) == "High Density") {
            rs_depth_sensor.set_option(RS2_OPTION_VISUAL_PRESET, i);
        }
    }
```
Which is copied straight from rs-measure example. Some notable remarks:
1. My program crashes 90% of the time. But not always. Also it seems to crash way less if I step through the code, or apply delays. Never made it to 0% though...
2. If I run the rs-measure the error does not occur. 
3. If I comment these lines, it often crashes on the next line:
`    depth_scale = rs_depth_sensor.get_option(RS2_OPTION_DEPTH_UNITS);`
With error 2:
```
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  hwmon command 0x2c failed. Error type:  (1).
```
4. This all seems to have started since the return of my vacation, after which the dkms package was upgraded to v2.22 (May have been on 2.21 or 2.20 before)
5. Before this code, I do all kinds of initialization stuff with the real-sense (like dynamically changing the exposure value). I also restart the pipeline because this part of the code needs the depth stream but others don't. It's difficult to isolate this all at the moment.
6. If I had to guess, something is wrong with rs_depth_sensor. This is made a little bit earlier in the function by:
`auto rs_depth_sensor = dev.first<rs2::depth_sensor>();`
dev a class global and is initialized somewhere else by a query_devices earlier in the program. Except for that I'm lost with this, I think something internally in the api may have gotten corrupted because all the stuff I do before this... Hints on what to look for much appreciated!

dorodnic, 2019-06-02:
This might be an unwanted side-effect of the 2.22 release. We are looking into it.

donghyunyoo, 2019-06-12:
Hello, 
I have the same issue. I made a capture program to get RGB and depth with libreasense. The version is 2.22. 
When I started my program, it crashed frequently. 
After I removed depth image capture, it got better, but it still crashes sometimes.
Do you have any timeframe to fix this issue?

Thanks.

kevindehecker, 2019-06-28:
I haven't had this problem anymore for a while, since I was upgraded to 2.23 and I upgraded the firmware to 05.11.06.250

kevindehecker, 2019-06-28:
Never mind I do have this problem still. It was just circumvented in another way.

kevindehecker, 2019-06-30:
Made some small progress. So it seems that if I change the order of things, I do not get this problem. 

Problem appears when I do this:
```
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_DEPTH, IMG_W, IMG_H, RS2_FORMAT_Z16, VIDEOFPS);
    if (hasIMU) {
        cfg.enable_stream(RS2_STREAM_ACCEL);
        cfg.enable_stream(RS2_STREAM_GYRO);
    }
    cam.start(cfg);

    auto rs_depth_sensor = dev.first<rs2::depth_sensor>();
    depth_scale = rs_depth_sensor.get_option(RS2_OPTION_DEPTH_UNITS); // ***CRASHES HERE***

    if (rs_depth_sensor.supports(RS2_OPTION_ENABLE_AUTO_EXPOSURE))
        rs_depth_sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1.0);
    if (rs_depth_sensor.supports(RS2_OPTION_EMITTER_ENABLED))
        rs_depth_sensor.set_option(RS2_OPTION_EMITTER_ENABLED, 1.f);

    if (rs_depth_sensor && rs_depth_sensor.is<rs2::depth_stereo_sensor>())
    {
        rs_depth_sensor.set_option(RS2_OPTION_VISUAL_PRESET, 
RS2_RS400_VISUAL_PRESET_HIGH_ACCURACY);                                   // ***OR CRASHES HERE***
    } else {
        throw my_exit("This is not a depth sensor");
    }

```
But not so if I start the cam *after*! :

    ```
    auto rs_depth_sensor = dev.first<rs2::depth_sensor>();
    depth_scale = rs_depth_sensor.get_option(RS2_OPTION_DEPTH_UNITS);

    if (rs_depth_sensor.supports(RS2_OPTION_ENABLE_AUTO_EXPOSURE))
        rs_depth_sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1.0);
    if (rs_depth_sensor.supports(RS2_OPTION_EMITTER_ENABLED))
        rs_depth_sensor.set_option(RS2_OPTION_EMITTER_ENABLED, 1.f);

    if (rs_depth_sensor && rs_depth_sensor.is<rs2::depth_stereo_sensor>())
    {
        rs_depth_sensor.set_option(RS2_OPTION_VISUAL_PRESET, RS2_RS400_VISUAL_PRESET_HIGH_ACCURACY);
    } else {
        throw my_exit("This is not a depth sensor");
    }

    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_DEPTH, IMG_W, IMG_H, RS2_FORMAT_Z16, VIDEOFPS);
    if (hasIMU) {
        cfg.enable_stream(RS2_STREAM_ACCEL);
        cfg.enable_stream(RS2_STREAM_GYRO);
    }
    cam.start(cfg);
```

doronhi, 2019-09-08:
The function `cam.start(cfg)` returns a profile through which you should take the device. 
The device (dev) you use in `dev.first<rs2::depth_sensor>()` is a different object then the one used by the pipeline.
Using both at the same time, as is the case when calling one after camera was started, seems like the cause for the crash.
You should fix your second version of the code accordingly since right now, the modifications you perform on `rs_depth_sensor` may not apply to the sensors running by the `cam.start(cfg)` command.

kevindehecker, 2019-09-13:
Aha, yes that does seem to fix the problem. Thanks!

Of course, as a user of the api another discussion is whether this is expected behavior. Should I expect the rs2::device itself to change whenever I change some settings in the configuration? 

A explicit bit of documentation would be helpful in any case I guess.

kevindehecker, 2019-10-01:
Since the error has been solved, I think it is OK to close the issue.

closed 2019-10-01 by kevindehecker