realsenseai/librealsense #4118: synchronization and frame of D435I

Garand0o0, 2019-05-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |   Linux (Ubuntu 16)| 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            | 2.0                           | 
| Language                            | C#                         | 
| Segment			|  Robot                           | 

### Issue Description
<Describe your issue / question / feature request / etc..>
My D435I can display on realsense-viewer.Then I want to select the fps of RGB camera.There are 6 15 30 60 on realsense-viewer.Does D435I only support those fps?Can I select other fps? I want to use Multiple sensors.But the sync button cannot be pressed.Does the realsense-viewer support sync?

Samvid95, 2019-06-03:
Probably Realsense viewer have the built in code for syncing multiple sensors. For your FPS question, I am not familiar with Realsense-viewer so I don't know.

ev-mp, 2019-06-10:
@Garand0o0 hello,
To list all the supported video formats, resolutions and fps rates you can use this utility [link](https://github.com/IntelRealSense/librealsense/issues/3416#issuecomment-471720605)
>sudo apt install v4l-utils
ls /dev/video* -1a | xargs -n1 v4l2-ctrl --list-formats-ext -d 

And afaik,  the RGB stream max fps rate is 60 indeed.
Regarding the syncer in `realsense-viewer` - it can be used for all sensors, but for D435i it is intentionally switched off to improve GUI responsiveness with high-rate Gyro and Accel streams (400 and 250 fps accordingly).

To get all D435i streams synchronized out of the box you can use the SDK's `pipe.wait_for_frames` API. Note that due to the ~20 factor in FPS rates between, say depth and Gyro, the syncer will synchronize and provide coherent sets of video (Depth/IR/RGB) streams, but the IMU streams will be served immediately, i.e. Gyro and Accel frames will always arrive in separate calls.

closed 2019-06-28 by RealSenseCustomerSupport