realsenseai/librealsense #4120: no matching constructor for initialization of 'std::thread'

chankamlam, 2019-05-30:
I am making in wrapper of openni2 and showing this error to me, how can I fix it?
[image]

RealSenseCustomerSupport, 2019-07-01:
----------------------------------------------

@chankamlam Sorry for late response. Did you still have the issue? Looking forward to your update. Thanks!

chankamlam, 2019-07-01:
> @chankamlam Sorry for late response. Did you still have the issue? Looking forward to your update. Thanks!

yes

RealSenseCustomerSupport, 2019-07-02:
----------------------------------------------

@chankamlam What's SDK version and Xcode version you use?

RealSenseCustomerSupport, 2019-07-08:
----------------------------------------------

@chankamlam Could you please update your SDK version and Xcode version when you get the issue? Thanks!

RealSenseCustomerSupport, 2019-07-11:
----------------------------------------------

@chankamlam Could you update? Thanks!

RealSenseCustomerSupport, 2019-07-15:
----------------------------------------------

@chankamlam pPease provide your SDK version and Xcode version if you still need support on this. Thanks!

RealSenseCustomerSupport, 2019-07-18:
----------------------------------------------

@chankamlam No response got from you yet. Will close the issue at this point.

closed 2019-07-18 by RealSenseCustomerSupport