realsenseai/librealsense #4121: Processing Rosbags in JavaScript

thomasliu623, 2019-05-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |   MacOS Mojave 10.14 | 
| SDK Version                            |  2.16.5                           | 
| Language                            |  C++                         | 

### Issue Description

Hi! I am looking to process librealsense's Rosbag outputs inside a React app, and was wondering if you had any recommendations for how to process Rosbag files in JavaScript? One option that I found is the [Cruise Automation Rosbag library](https://github.com/cruise-automation/rosbag.js), but was wondering if you've had success with applying that to librealsense's Rosbag outputs? Or perhaps do you know of any wrapper or library that allows use of the Librealsense library in JavaScript?

RealSenseCustomerSupport, 2019-08-27:
----------------------------------------------

For playback of recorded files functionality in JavaScript, please take a look at Librealsense Node.js wrapper:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs

RealSenseCustomerSupport, 2019-09-03:
----------------------------------------------

Hi Thomas,

Do you need any further clarifications regarding this topic?

closed 2019-09-10 by RealSenseCustomerSupport