realsenseai/librealsense #4125: 'RS2_OPTION_HISTOGRAM_EQUALIZATION_ENABLED'  not defined in Python?

Marc3636, 2019-05-31:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |   Linux Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

Hi,

I want to disable the Histogram_Equalization, so that I can get the same depth image in a python script like I get in the Realsense Viewer right now. I can set the min and max distances with set_option but I cannot set the RS2_OPTION_HISTOGRAM_EQUALIZATION_ENABLED in Python. I get a message "not defined" (see below). What can I do so that I can disable/enable HISTOGRAM_EQUALIZATION in my Python script?

Here is part of the code:
colorMap = rs.colorizer()
colorMap.set_option(rs.option.min_distance,0)
colorMap.set_option(rs.option.max_distance,6)
print("colorizer - min_distance: %f" %(colorMap.get_option(rs.option.min_distance)))
print("colorizer - max_distance: %f" %(colorMap.get_option(rs.option.max_distance)))
colorMap.set_option(RS2_OPTION_HISTOGRAM_EQUALIZATION_ENABLED,0)

And the result is this:
colorizer - min_distance: 0.000000
colorizer - max_distance: 6.000000

---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
<ipython-input-2-39f5ef7c2451> in <module>
     14 print("colorizer - min_distance: %f" %(colorMap.get_option(rs.option.min_distance)))
     15 print("colorizer - max_distance: %f" %(colorMap.get_option(rs.option.max_distance)))
---> 16 colorMap.set_option(RS2_OPTION_HISTOGRAM_EQUALIZATION_ENABLED,0)
     17 
     18 depth_scale = depth_sensor.get_depth_scale()

NameError: name 'RS2_OPTION_HISTOGRAM_EQUALIZATION_ENABLED' is not defined

dorodnic, 2019-06-02:
In python that would be `rs.option.histogram_equalization_enabled`

Marc3636, 2019-06-03:
Thank you dorodnic, now it works perfectly;)

closed 2019-06-03 by Marc3636