realsenseai/librealsense #4127: How to convert Fisheye frame to cv::mat?

Samvid95, 2019-05-31:
I am trying to convert the fisheye frame into the Mat but I am not getting any results back. I don't know what I am missing here. Can someone please check it out? 

```
namedWindow("T265", 1);
while (true) {
rs2::frameset data = pipe.wait_for_frames();

auto fisheye = data.first_or_default(RS2_STREAM_FISHEYE);
auto vf = fisheye.as<rs2::video_frame>();
const int w = vf.get_width();
const int h = vf.get_height();
cv::Mat fisheye_MAT = Mat(Size(w, h), CV_16UC1, (void*)fisheye.get_data(), Mat::AUTO_STEP);
imshow("T265", fisheye_MAT);
}
```

dorodnic, 2019-06-02:
Maybe `CV_16UC1` is causing the problem? Fisheye is single plane one byte per pixel gray-scale image.

Samvid95, 2019-06-03:
Yup I am also pretty sure that `CV_16UC1` is causing this problem but I am unable to figure out what can I use instead of that, as I have also tried `CV_8UC1`, `CV_32FC1` but none of them seems to work and at this point I am brute forcing this things so I am getting more cofused.

Samvid95, 2019-06-03:
Also the Fisheye frame data is of `Y8` format and usually we use `CV_8UC1` for that but it is also not working in that case.

dorian3d, 2019-06-18:
Hi @Samvid95 . I'm sure `CV_8UC1` is the correct type for `cv::Mat`.
You may be missing a line like `waitKey(30)` after `imshow` to allow OpenCV draw the image and handle the window events.

RealSenseCustomerSupport, 2019-07-11:
----------------------------------------------

@Samvid95 Did you get it through with dorian3d's suggestion? Any other questions? Looking forward to your update. Thanks!

###

RealSenseCustomerSupport, 2019-07-15:
----------------------------------------------

@Samvid95 Could you please update? Thanks!

Samvid95, 2019-07-15:
Yes I got it to work. @dorian3d  and @RealSenseCustomerSupport  I have the new solution and I'll post the update pretty soon. I checked I had the `imshow ` and `waitKey(30)` but the way you enable the camera is slightly different. Let me post the solution tomorrow so someone might get some help.

RealSenseCustomerSupport, 2019-07-15:
----------------------------------------------

@Samvid95 Glad to know that and thanks for sharing! Looking forward to your post. Thanks!

Samvid95, 2019-07-15:
`
// Declare RealSense pipeline, encapsulating the actual device and sensors 

rs2::pipeline pipe;
 // Create a configuration for configuring the pipeline with a non default profile

 rs2::config cfg; 

// Enable both image streams // Note: It is not currently possible to enable only one 

cfg.enable_stream(RS2_STREAM_FISHEYE, 1, RS2_FORMAT_Y8); 
cfg.enable_stream(RS2_STREAM_FISHEYE, 2, RS2_FORMAT_Y8); 

// Start pipeline with chosen configuration
 pipe.start(cfg);  
 using namespace cv; 
const auto window_name = "Display Image"; 
namedWindow(window_name, WINDOW_AUTOSIZE);    
 while (waitKey(1) < 0 && getWindowProperty(window_name, WND_PROP_AUTOSIZE) >= 0) { 

    rs2::frameset data = pipe.wait_for_frames(); 
    rs2::video_frame fisheye_frame = data.get_fisheye_frame(2);   
    // Query frame size (width and height) 
    const int w = fisheye_frame.as<rs2::video_frame>().get_width(); 
    const int h = fisheye_frame.as<rs2::video_frame>().get_height(); 
    cv::Mat fisheye_mat = Mat(Size(w, h), CV_8UC1, (void *)fisheye_frame.get_data(), Mat::AUTO_STEP); 
    imshow("T265", fisheye_mat);   
 }`

RealSenseCustomerSupport, 2019-07-17:
----------------------------------------------

@Samvid95 Thanks again! Will close this ticket accordingly.

closed 2019-07-17 by RealSenseCustomerSupport