realsenseai/librealsense #4128: Depth Frame not Showing all Metadata Options

derens99, 2019-05-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Camera Model: 2 D435's
Firmware Version: 05.10.13.00
OS: Ubuntu 18.04
Platform : PC
SDK: 2.0
Language: C++
Segment: Depth and RGB Capture/Recording

Hi,

I am trying to get all the metadata from a depth frame, and in the docs it says there is supposed to be these listed in the output:

FRAME_COUNTER
FRAME_TIMESTAMP
SENSOR_TIMESTAMP
ACTUAL_EXPOSURE 
GAIN_LEVEL 
AUTO_EXPOSURE 
TIME_OF_ARRIVAL
BACKEND_TIMESTAMP 
ACTUAL_FPS 

But, when I test my code and tets rs-save-to-disk, the only metadata values I am getting back from the camera are:
FRAME_TIMESTAMP
TIME_OF_ARRIVAL
BACKEND_TIMESTAMP
ACTUAL FPS

I at least need the Sensor Timestamp in order to check for my hardware trigger I am trying to implement in my cameras.

Thanks

dorodnic, 2019-06-02:
This implies not all kernel patches have been applied. Try installing `librealsense2-dkms` as described here: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

derens99, 2019-06-03:
Hi. Yes, you're right. I tried it on my Ubuntu 16.04 version, and it dumped out all the metadata values instead of 4. Thanks.

closed 2019-06-03 by derens99