realsenseai/librealsense #4131: Calibration parameters are suddenly back to gold/factory

matspetter, 2019-06-01:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.10.6.0 | 
| Operating System & Version | Linux Ubuntu 16 | 
| Kernel Version (Linux Only)    | 4.4.0.145-generic | 
| Platform                                 | Intel NUC |
| SDK Version                            | 2.10.0 | 
| Language                            | c++ | 
| Segment			| Robot | 
|---------------------------------|------------------------------------------- |

We are manually calibrating our D435 with the Intel.Realsense.DynamicCalibrator . After a successful calibration we save the new calibration params to file with "CustomRW -r -f file.xml". All this works fine, the accuracy is good etc.

**Sporadically though these calibration parameters in the cam seems to be reset back to gold/factory values for no obvious reason.**

When that happens we can restore to our saved calibration params again with "CustomRW -w -f file.xml" and the cam works fine again (for some time).

We have seen that there is a DSO-10002 that seems to target this problem but according to the "Specification Update" (https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense-D400-Series-Spec-Update.pdf) this should have been corrected in fw 5.10.3 but we are using 5.10.6 so we should be on the safe side or!!??

Btw, are those DSOs public somewhere? I can only find references to them in the Spec Update but I cannot find the "source".

RealSenseCustomerSupport, 2019-07-01:
----------------------------------------------

@matspetter Yes. FW5.10.6 should fix the problem. Or else you can also upgrade FW to the latest v5.11.6.250 which can be got from https://downloadmirror.intel.com/28870/eng/D400_Series_Production_FW_5_11_6_250.zip
 DSOs are in our internal tracking system so they're not public.

RealSenseCustomerSupport, 2019-07-05:
----------------------------------------------

@matspetter Any  other questions about this ticket? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2019-07-09:
----------------------------------------------

@matspetter If no other questions about this issue, will close it accordingly. Thanks!

closed 2019-07-09 by RealSenseCustomerSupport