realsenseai/librealsense #4136: High CPU load when using D435

jinxzzz, 2019-06-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win7  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C#}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
When I call the RGB image and depth image of the D435 under Winform, my CPU load reaches 90%, which makes it impossible for me to run other programs.Is there any solution? My platform is Windows 7 and my CPU is I3 3210.
Here's my code：
        public void depth_image()
        {
            var pipeline = new Pipeline();
            var colorizer = new Colorizer();
            var cfg = new Config();
            cfg.EnableStream(Intel.RealSense.Stream.Depth, 640, 480);
           // cfg.EnableStream(Intel.RealSense.Stream.Color, 640, 480, Format.Bgr8, 30);
            var pp = pipeline.Start(cfg);

            while (true)
            {
                using (var frames = pipeline.WaitForFrames())
                using (var depth = frames.First(x => x.Profile.Stream == Intel.RealSense.Stream.Depth) as DepthFrame)
                //using(var DepthStream = pp.GetStream(Intel.RealSense.Stream.Depth) as VideoStreamProfile)
                {
                   
                    DepthFrame depthFrame = frames.DepthFrame.DisposeWith(frames) as DepthFrame;
                    VideoFrame colorizedDepth = colorizer.Process(depthFrame).DisposeWith(frames) as VideoFrame;
                    Bitmap bmpDepth = ToBitmap(colorizedDepth);
                  
                    Bitmap temp2 = new Bitmap(bmpDepth);
                    Bitmap DepthImg = new Bitmap(640, 480, depthFrame.Stride, PixelFormat.Format24bppRgb, depthFrame.Data);
                    pictureBox1.InitialImage = null;
                    pictureBox2.InitialImage = null;
                    //pictureBox1.Image = temp1;
                    pictureBox2.Image = temp2;
                    pictureBox3.Image = DepthImg;
                    //pictureBox4.Image = temp1;
                    //pictureBox1.Image.Save("Picturebox.jpg", ImageFormat.Jpeg);
                    //bmpColor.Dispose();
                    bmpDepth.Dispose();
                    //temp.Dispose();
                    //pictureBox1.Image.Dispose();
                   // colorFrame.Dispose();
                    depthFrame.Dispose();
                    colorizedDepth.Dispose();

                }//end using
            }//end while    
        }

dorodnic, 2019-06-03:
Are you getting similar CPU utilization when using [official RealSense Viewer](https://github.com/IntelRealSense/librealsense/releases/download/v2.22.0/Intel.RealSense.Viewer.exe)?

jinxzzz, 2019-06-06:
> 使用[官方RealSense Viewer](https://github.com/IntelRealSense/librealsense/releases/download/v2.22.0/Intel.RealSense.Viewer.exe)时，您是否获得类似的CPU利用率？
Yes, when I add System. Sleep (500) to the while loop, the CPU usage drops to 60 percent

RealSenseCustomerSupport, 2019-06-26:
----------------------------------------------

@jinxzzz For RealSense Viewer, did you get it by installing Win7 SDK? Or you build it by yourself? If you build it by yourself, please make sure you build Release version instead of Debug version. Release build will get lower CPU compared with Debug build.

RealSenseCustomerSupport, 2019-07-08:
----------------------------------------------

@jinxzzz If no other questions will closet this ticket accordingly. Thanks!

closed 2019-07-08 by RealSenseCustomerSupport