realsenseai/librealsense #4137: how can I get the value in Z-axis( the vertical distance from the point to the X-Y plane)?

ChenJfChen, 2019-06-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win10}  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC.  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python 2.7 }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
   The distance got by the function depth.get_distance() seems to be the distance from one point to the camera, then how can I get the value in Z-axis( the vertical distance from the point to the X-Y plane)? Is there any existed function in the API to get this or only using the math formula x2+y2+z2=distance2?

dorodnic, 2019-06-03:
Hi @ChenJfChen 
The distance you get from get_distance is from the X-Y plane.

RealSenseCustomerSupport, 2019-06-24:
----------------------------------------------

Hi @ChenJfChen
Do you still have question on this issue?

ChenJfChen, 2019-06-25:
I'm so sorry to forget to close this issue

closed 2019-06-25 by ChenJfChen

PointCloudNiphon, 2020-04-11:
> Hi @ChenJfChen
> The distance you get from get_distance is from the X-Y plane.

could you please provide me with the document that show the get_distance giving "The distance you get from get_distance is from the X-Y plane."

PointCloudNiphon, 2020-04-11:
> > Hi @ChenJfChen
> > The distance you get from get_distance is from the X-Y plane.
> 
> could you please provide me with the document that show the get_distance giving "The distance you get from get_distance is from the X-Y plane."

I find this confused example in Intel RealSense SDK 2.0\samples\hello-world\main.cpp
```C++
      // Query the distance from the camera to the object in the center of the image
        float dist_to_center = depth.get_distance(width / 2, height / 2);
```

PointCloudNiphon, 2020-04-11:
@RealSenseCustomerSupport