realsenseai/librealsense #4138: Change stream resolution failed with OpenNI2 wrapper [windows x64]

areha001, 2019-06-03:
Is there any one succeed to open the depth stream with another resolution instead of default VGA 640x480 by using wrapper OPENNI2 ?
I've compiled the rs2driver and got the rs2driver.dll ,and put it into the redist driverfolder and also the  folder  ${openni2}/tools/openni2/drivers/
I run the NiViewer.exe success and I got the left-depth-right-color image , according to the console , I'm using 640x480@30fps default preset resolution
But I can't change the resolution of  the depth and also the color stream. when I tried to change it , the screen would be frozen and I can only exit the program
[image]
Just here, I can not to change the resolution as the guider says.

It is the same I'm tring to compile the source code sample of the openni2/nite2 . I can not init the stream with another resolution
`

	openni::VideoStream mDepthStream;
	mDepthStream.create(m_device, openni::SENSOR_DEPTH);
	
	//create 1280x720@30fps depth stream
	openni::VideoMode mModeDepth;
	mModeDepth.setResolution(1280, 720);
	mModeDepth.setFps(30);
	mModeDepth.setPixelFormat(openni::PIXEL_FORMAT_DEPTH_1_MM);
	mDepthStream.setVideoMode(mModeDepth);
	mDepthStream.start();

	nite::NiTE::initialize();

	
	nite::Status status = m_pUserTracker->create();  //failed , not OK
	if (status != nite::STATUS_OK)
	{
		return openni::STATUS_ERROR;
	}
	

`
Is there anyone succeed to change the resolution instead of VGA here? with OpenNI2&NITE2

aaranta, 2020-01-07:
> Is there any one succeed to open the depth stream with another resolution instead of default VGA 640x480 by using wrapper OPENNI2 ?
..
> Is there anyone succeed to change the resolution instead of VGA here? with OpenNI2&NITE2

@areha001 you have to turn off your depth, color and IR streams first. Then only you change to your desired resolution and turn on all the streams again.
I've tried 1280x720 @ 30fps for depth, color and IR streams in NiViewer. I can confirm it works for RealSense D415.

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@aaranta Thanks for your share!
@areha001 Did you try what @aaranta sugested? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@aaranta Did you get chance to try? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2020-02-25:
----------------------------------------------

@aaranta Will close this if no other questions. Thanks!

closed 2020-02-25 by RealSenseCustomerSupport

mediadog, 2021-03-29:
I am having a similar problem.  When I change the depth and color resolutions to 848x480 in a program using OpenNI2 (stopping and starting the streams with a full OpenNI2 restart), the resulting images are scrambled like the row/column indexing is wrong.  1280x720 works fine, so it's not just an aspect ratio issue.

What I get at 640x480:
[image]

and 848x480:
[image]

This is with both a D435 and D455, on both Win7 and Win10 with the drivers for each built from 2.42.0.