realsenseai/librealsense #4139: Method to plot Position Data of t265 in 2D / 3D . Issue/Help :

TanmayPathak, 2019-06-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Operating System & Version |   Ubuntu 16  | 
| Platform                                 | Intel NUC  |
| SDK Version                            |  2.22                         | 
| Language                            |  python                          | 
| Segment			|  Robot                        | 

### Issue Description
Any method or technique available for 2D / 3D Graph / Map plotting with python using  Intel RealSense T265 . Similar to intel RealSense Viewer output.

Samvid95, 2019-06-03:
I am not to familiar with the Python API but in C++ we do this with the integration with OpenGL. You should check out that route where you can use openGL and plot by yourself.

millsrandy, 2019-06-07:
I'm also interested in doing this, so if you find a solution please let me know.  So far, I have everything connected in Unity and am able to record .bag files with Unity (Windows 10), I just can't figure out how to convert the .bag data to "3D" data that I can export as fbx and use in other 3D software (like C4D).

Samvid95, 2019-06-07:
I think T265 is really least developed product from there yet. I already have couple of other issues open on T265.

JimmyFroost, 2019-06-21:
You should follow [this example](
https://github.com/IntelRealSense/librealsense/tree/master/examples/trajectory).
That's what I used to get X Y Z coordinates and then be able to plot the trajectory, one point at a time.

closed 2019-07-01 by TanmayPathak