realsenseai/librealsense #4140: Auto exposure below 60 faulty

nicovanduijn, 2019-06-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.6.200 | 
| Operating System & Version |   Linux (Ubuntu 16.04LTS)  | 
| Kernel Version (Linux Only)    | 4.15.0-50-generic        | 
| Platform                                 | PC  |
| SDK Version                            | 2.22.0 }                          | 
| Segment			|  Robot                     | 

### Issue Description
When using the D435 through USB2.1 with Firmware 5.11.6.200 with the depth stream at 15Hz, the camera produces faulty IR images and depth images for exposure values below 61. Similar for auto-exposure, once the auto exposure corrects to a value below 61, the same image appears. 

Note 1: This does not happen when using USB3.1
Note 2: This does not happen when the frame rate is set to 30Hz

**Exposure manually set to 61**
[image]

**Exposure 60 or below, or auto-exposure in very bright light**
[image]

RealSenseCustomerSupport, 2019-06-24:
----------------------------------------------

Hi

Thanks a lot for reporting.

Yes, that's the known issue (https://github.com/IntelRealSense/librealsense/issues/4022) and will be fixed in firmware 5.11.12.0.

Thank you

RealSenseCustomerSupport, 2019-07-01:
----------------------------------------------

Just an update: the bug will be fixed within firmware 5.11.13.0.

Thanks

RealSenseCustomerSupport, 2019-07-11:
----------------------------------------------

This issue is just fixed in FW5.11.12.0. Please kindly wait for the official release.

closed 2019-07-12 by RealSenseCustomerSupport