realsenseai/librealsense #4142: distance_to_object.ipynb outdated instructions

kat-mcg-lin, 2019-06-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i and T365 | 
| Firmware Version                   | 05.11.04.00 | 
| Operating System & Version |   Linux - Ubuntu 18.04 | 
| Kernel Version     |  4.9.140                                      | 
| Platform                                 | Jetson Nano |
| SDK Version                            |  2.0 (I think)                     | 
| Language                            |  python                         | 
| Segment			| Robot                      | 

### Issue Description
I am trying to follow the step by step code about calculating distances to recongized objects (https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb) and the link that describes how to get Mobile Net SSD seems to no longer be up to date, as when I follow the instructions one of the websites no longer exists so I can't get the right files to use the Mobile Net

[image]

How can I complete this code example? Any other method to use the MobileNet SSD, I tried going directly to the git source and it seems waaay more complicated and goes over my head.

thanks!

lramati, 2019-09-02:
It seems like you're using a pretty old version of the notebooks. I couldn't reproduce your issue, but I just opened #4787 to try and fix some other issues. Could you see if the following binder link to my fork works for you?

[image]

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

Hi @kat-mcg-lin,

Any update? If the issue resolved, will close it.

closed 2019-11-01 by RealSenseCustomerSupport