realsenseai/librealsense #4143: Color Information About Each Pixel

NickolasCavagnaro, 2019-06-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 5.11.4 | 
| Operating System & Version |   { Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |   4.15.0                                        | 
| Platform                                 | PC  |
| SDK Version                            |  {  2.22.0 }                          | 
| Language                            |  { nodejs }                          | 
| Segment			|  { others }                          | 

### Issue Description
I am trying to acquire the R,G,B values for every pixel so I can generate statistics like the average of each color throughout a region of interest, the respective standard deviations, etc. Right now, I am able to capture colorized depth maps and the RGB camera, but have not yet encountered a resource detailing how to acquire per-pixel color information.

MartyG-RealSense, 2019-06-03:
The link below has a sample of scripting for getting the RGB value of a specific pixel.

https://github.com/IntelRealSense/librealsense/issues/3364

NickolasCavagnaro, 2019-06-04:
I'm confused about this line of code:

auto stride = rgb.as<rs2::video_frame>().stride();

Also, the sample was in C++. How would this work with the node.js wrapper?

closed 2019-06-07 by NickolasCavagnaro