realsenseai/librealsense #4144: long distance (3.5m~6.5m)  a great many black hole

xiaoyanluo, 2019-06-04:
| Camera Model                      D435i
| Firmware Version                  5_11_1_100
| Operating System & Version  Wi10
| Kernel Version (Linux Only)    (e.g. 4.14.13)                                     
| Platform                                 | PC
| SDK Version                            | Intel RealSense SDK 2.0                          | 
| Language                            |  C                   | 
| Segment			|  others }                          | 

### Issue Description
hi :       when i work in  long distance (3.5m~6.5m)   ,a great many black holes . I have changed the depth units scale ,but it is bad too.
Thanks in advance!

xiaoyanluo, 2019-06-04:
[image]

xiaoyanluo, 2019-06-04:
[image]

RealSenseCustomerSupport, 2019-06-25:
----------------------------------------------

Hi,
Could you please attach a left IR image along with depth image to show the problem?
Did you try to increase the projector power?

xiaoyanluo, 2019-06-25:
Thanks for your responses.

closed 2019-06-25 by xiaoyanluo