realsenseai/librealsense #4145: Realsense SDK crashes (inside ROS driver) while viewer works

fgrzadkowski, 2019-06-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |   Linux (Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.4.0-131-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2.22.0                          | 
| Language                            |  C                         | 
| Segment			|  Robot }                          | 

### Issue Description
I'm running realsense ROS drivers in version 2.2.4 with SDK 2.22.0. I have 3 RS cameras, each of them using separate USB controller (on a PCI card with 4 slots).  I found out that when I plug one of the cameras to port A then I get fatal errors from librealsense (see below), while when I plug it into another port everything works. Normally I would think this is a hardware issue, but realsense-viewer is able to handle this camera on both USB ports. Is it some bug in librealsense? ROS driver? Something else?

Error:
```
[ INFO] [1559630492.383514741]: Running with LibRealSense v2.22.0
 04/06 06:41:33,445 ERROR [139735912888064] (types.h:243) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 04/06 06:41:33,546 WARNING [139735912888064] (rs.cpp:279) null pointer passed for argument "device"
terminate called after throwing an instance of 'rs2::backend_error'
  what():  set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
```

RealSenseCustomerSupport, 2019-07-02:
----------------------------------------------

@fgrzadkowski Did you get the issue with ROS when plug the camera to Port A for every time? Is it USB2 or USB3 port?

RealSenseCustomerSupport, 2019-07-08:
----------------------------------------------

@fgrzadkowski Could you please update if you still need support on this? Thanks!

fgrzadkowski, 2019-07-08:
Sorry for not updating it. I missed previous notification. 

All ports are USB3. This was very repeatable (I don't remember if it was exactly every single time, but it was probably more than 90% of times. Since then I've changed the PCI card and I haven't seen this issue, but it'd be still to understand why librealsense is crashing while realsense-viewer can handle those errors.

closed 2019-08-16 by RealSenseCustomerSupport