realsenseai/librealsense #4146: Matlab T265 Fisheye view

JimmyFroost, 2019-06-04:
| Camera Model |T265} |
| Operating System  | Wind 10 |
| Platform | PC |
| SDK Version | 2.20 |
| Language | Matlab 2019 |

Hello, since I've got no answer since last week on  a closed issue, I am reposting my message to   @RealSenseCustomerSupport ,
 
I've been trying to continue getting the live image from fisheye cameras, accordingly to this [CPP code](https://github.com/IntelRealSense/librealsense/blob/master/examples/ar-basic/rs-ar-basic.cpp).

On line 107, there is no equivalent for **get_fisheye_frame** in Matlab (yet), so I tried to code it on my own since the function is very similar to **get_infrared_frame,** based on lines 945 to 980 in [this code file](https://github.com/IntelRealSense/librealsense/blob/f361e56f8a2fb5093a1d663293421c92dd432bba/include/librealsense2/hpp/rs_frame.hpp).

So here it is in **frameset.m** : 
```

  function fisheye_frame = get_fisheye_frame(this, index)
            if (nargin == 1)
                ret = realsense.librealsense_mex('rs2::frameset', 'get_fisheye_frame', this.objectHandle);
            else
                validateattributes(index, {'numeric'}, {'scalar', 'real', 'integer'}, '', 'index', 2);
                ret = realsense.librealsense_mex('rs2::frameset', 'get_fisheye_frame', this.objectHandle, int64(index));
            end
            fisheye_frame = realsense.video_frame(ret);
        end
```

Evnetually, I get this error :

> Error using librealsense_mex
> Unknown Command received.
> 
> Error in realsense.frameset/get_fisheye_frame (line 56)
>                 ret = realsense.librealsense_mex('rs2::frameset', 'get_fisheye_frame', this.objectHandle, int64(index));
>
> Error in T265_Stream (line 67)
> img_fish =  fs.get_fisheye_frame();

Is there something I can do or am I doomed to wait for an update of yours ?

Thanks.

lramati, 2019-06-13:
The matlab wrapper is comprised of two components. One is the glut of .m files which exposes the functionality in a more convenient and matlab-esque manner, and the other is the actual mex file that translates the calls from matlab code to C++ code. You would need to also update the mex file's source at librealsense_mex.cpp and recompile it.
I will release a PR for this functionality soon if you would rather wait for an official implementation, but that would still require manually compiling from source until it gets merged and becomes part of the official release

JimmyFroost, 2019-06-14:
Hello @lramati 

Thanks for your informations. I will gladly wait for your official implementation 👍 
If you can provide me with the way to compile manually from source, I will do my best to achieve it.

Have a nice week end.

lramati, 2019-06-14:
The readme file in the Matlab folder has all the instructions you'll need.
In theory it should work on any OS, but at the moment we only verify that
it works on Windows

JimmyFroost, 2019-06-14:
alright I'll have a look when necessary.
Thank you

RealSenseCustomerSupport, 2019-07-04:
----------------------------------------------

Fixed in librealsense 2.24.0.

closed 2019-07-04 by RealSenseCustomerSupport

JimmyFroost, 2019-07-04:
thank you, I managed to integrate both fisheyes stream. 👍