realsenseai/librealsense #4147: Calibration tool

EmmanuelCos, 2019-06-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 

Hello,
I would like to know if it's possible to go back to the last calibration data of the camera, after the calibration has been done with the calibration tool. Thas is to say is there any way to cancel a calibration done with calibration tool?
Thank you,
Emmanuel

MartyG-RealSense, 2019-06-04:
I'm not aware of an 'Undo' function on the Custom Calibrator software.  One approach to restore the previous state though would be to write the current calibration to a file with the CustomRW tool before you make changes and then load that file back in if you need to restore the previous settings.  

Pages 46 to 51 of the user guide for the calibrator software describe the process of writing and reading calibration files with the CustomRW tool.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

RealSenseCustomerSupport, 2019-06-28:
----------------------------------------------

@EmmanuelCos Did you get chance to try as MartyG-RealSense suggested? If you want to go back to factory golden calibration data, you can also try "Intel.Realsense.CustomRW -g".

RealSenseCustomerSupport, 2019-07-08:
----------------------------------------------

@EmmanuelCos If no other questions about this ticket, will close it accordingly. Thanks!

closed 2019-07-08 by RealSenseCustomerSupport