realsenseai/librealsense #4148: How can I recover realsense devices stuck in DFU mode?

xeonxu, 2019-06-04:
As the title says, I find that there is no exiting DFU function in intel-realsense-dfu linux version. But windows version has. It's a little bit inconvenient for me.  Once I try to read device firmware version using intel-realsense-dfu, device will always stuck in DFU mode. So I can't use the camera unless re-plug it. 
Could you please add a parameter for intel-realsense-dfu in case to exit from DFU mode?

xeonxu, 2019-06-17:
@seebq @wspr @martinus @chachi Any body here can help me?

RealSenseCustomerSupport, 2019-07-09:
----------------------------------------------

Hi

Yes, that's the known issue. Can you please check the firmware by using LibRealSense?

Thanks

closed 2019-07-23 by RealSenseCustomerSupport