realsenseai/librealsense #4154: How to convert pixel coordinates to 3d spatial coordinates？

jinxzzz, 2019-06-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C# }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I used C# to call out the RGB image and depth image of D435, which were not aligned, and the resolution is 640×480. The pixel coordinates of the geometric center of the target object are identified in the RGB image, and the pixel coordinates of the corresponding depth image are obtained through corresponding conversion. How to convert the pixel coordinates into 3D spatial coordinates?

RealSenseCustomerSupport, 2019-07-02:
----------------------------------------------

@jinxzzz Please check if below code snippet helps.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/helper_functions.py#L121-L147

RealSenseCustomerSupport, 2019-07-05:
----------------------------------------------

@jinxzzz Did you get chance to try? Any other questions about this ticket? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-07-11:
----------------------------------------------

@jinxzzz If no other questions will close this ticket. Thanks!

closed 2019-07-11 by RealSenseCustomerSupport