realsenseai/librealsense #4156: intel-realsense-dfu missing a dependency

paulbovbel, 2019-06-05:
The following log on a minimal system leads me to believe intel-realsense-dfu is missing a dependency that would pull in the udevadm command.

```

[2019-06-04T23:29:40.437Z] Preparing to unpack .../intel-realsense-dfu_1.1.0_amd64.deb ...

[2019-06-04T23:29:40.437Z] Unpacking intel-realsense-dfu (1.1.0) ...

[2019-06-04T23:29:40.437Z] Setting up intel-realsense-dfu (1.1.0) ...

[2019-06-04T23:29:40.437Z] updating udev rules

[2019-06-04T23:29:40.437Z] /var/lib/dpkg/info/intel-realsense-dfu.postinst: 24: /var/lib/dpkg/info/intel-realsense-dfu.postinst: udevadm: not found

[2019-06-04T23:29:40.437Z] dpkg: error processing package intel-realsense-dfu (--configure):

[2019-06-04T23:29:40.437Z]  subprocess installed post-installation script returned error exit status 127

[2019-06-04T23:29:40.437Z] Errors were encountered while processing:

[2019-06-04T23:29:40.437Z]  intel-realsense-dfu

[2019-06-04T23:29:40.437Z] E: Sub-process /usr/bin/dpkg returned an error code (1)

[2019-06-04T23:29:40.437Z] ERROR: the following rosdeps failed to install

[2019-06-04T23:29:40.437Z]   apt: command [sudo -H apt-get install -y intel-realsense-dfu] failed

[2019-06-04T23:29:40.437Z] executing command [sudo -H apt-get install -y intel-realsense-dfu]

```

paulbovbel, 2019-06-05:
Pre-installing udev in a docker container leads to:

```
[2019-06-05T21:04:42.330Z] Setting up udev (237-3ubuntu10.21) ...

[2019-06-05T21:04:42.585Z] A chroot environment has been detected, udev not started.

...

[2019-06-05T21:04:43.352Z] Setting up intel-realsense-dfu (1.1.0) ...

[2019-06-05T21:04:43.352Z] updating udev rules

[2019-06-05T21:04:43.352Z] dpkg: error processing package intel-realsense-dfu (--configure):

[2019-06-05T21:04:43.352Z]  installed intel-realsense-dfu package post-installation script subprocess returned error exit status 2
```

I'm guessing that the intel-realsense-dfu postinst script should be tolerant to the absence of udev...

RealSenseCustomerSupport, 2019-08-26:
----------------------------------------------

@shadowdk3 Thanks for raising this! Now we have one new firmware upgrade tool https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update
Could you please try this new version to see if any issue? Thanks!

closed 2019-08-26 by paulbovbel