realsenseai/librealsense #4157: realsense-viewer.cpp mixes up argv and argc

eholly1, 2019-06-05:
Y'all should switch these.

[argv](https://github.com/IntelRealSense/librealsense/blob/7e833e6929677ae1f8fc7c64bf4961de1f4b6f52/tools/realsense-viewer/realsense-viewer.cpp#L255)

[argc]
(https://github.com/IntelRealSense/librealsense/blob/7e833e6929677ae1f8fc7c64bf4961de1f4b6f52/tools/realsense-viewer/realsense-viewer.cpp#L259)

ev-mp, 2019-06-06:
@eholly1, thanks for pointing this out.
The reason that it works is that the params are actually swapped in the `main`:
>int main(int argv, const char** argc) try

We'll adjust it

RealSenseCustomerSupport, 2019-07-02:
----------------------------------------------

@eholly1 Thanks for your input. The fix is now in v2.24. https://github.com/IntelRealSense/librealsense/blob/v2.24.0/tools/realsense-viewer/realsense-viewer.cpp#L245

closed 2019-07-02 by RealSenseCustomerSupport