realsenseai/librealsense #4158: rs-multicam sometimes upon startup

kroq-gar78, 2019-06-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |  Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  4.15.0-50-generic                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  2.22.0-0~realsense0.1106                          | 
| Language                            |  C++                          | 
| Segment			|  Multi-camera                          | 

### Issue Description
I am trying to run the example `rs-multicam` with three D435's (without hardware sync). Most of the time, it crashes with the error:

```
RealSense error calling rs2_pipeline_start_with_config(pipe:0x563c0c05c520, config:0x563c0c05c540):
    hwmon command 0x10 failed. Error type:  (1).
```

Other times, it crashes with this error:
```
RealSense error calling rs2_pipeline_start_with_config(pipe:0x55d55bb5e0b0, config:0x55d55bb5e0d0):
    hwmon command 0x30 failed. Error type:  (1).
```

Sometimes, it does run successfully and show the 6 streams (depth and color from each camera). I don't recall this happening with SDK version 2.18 (the last version I used). The crashes become less frequent if I use two cameras, and apparently non-existent if I only use one. The RealSense Viewer can stream depth+color from all cameras simultaneously, without error.

derens99, 2019-06-17:
Hello, I am receiving this error too. The exact same hwmon error. Have you found a fix for this?

kroq-gar78, 2019-06-22:
I didn't, but my workaround is to use `rs2::sensor` objects directly instead of using `rs2::pipeline`. I can post my `rs-multicam` example using sensors if you want.

derens99, 2019-06-24:
That would be very helpful if you could post your example. Thanks.

cga-cmu, 2019-06-24:
Same problem here in Intel NUC running realsense packages v. 2.23
Please post working example.
uname -a
Linux nuc0 4.18.0-24-generic #25~18.04.1-Ubuntu SMP Thu Jun 20 11:13:08 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

cga-cmu, 2019-06-25:
More details:
4 D435 cameras running Signed_Image_UVC_5_11_6_250.bin firmware
librealsense:development as of today Mon Jun 24 20:37:36 EDT 2019
uname -a
Linux a1 4.15.0-52-generic #56~16.04.1-Ubuntu SMP Thu Jun 6 12:03:31 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

cga-cmu, 2019-06-25:
several launches of rs-multicam. ^C marks successful launches

cga@a1:~/opencv-rs2$ ./rs-multicam
RealSense error calling rs2_pipeline_start_with_config(pipe:0x21f1d00, config:0x21ee7c0):
    hwmon command 0x10 failed. Error type:  (1).
cga@a1:~/opencv-rs2$ ./rs-multicam
RealSense error calling rs2_pipeline_start_with_config(pipe:0x1c081f0, config:0x1c083a0):
    hwmon command 0x30 failed. Error type:  (1).
cga@a1:~/opencv-rs2$ ./rs-multicam
RealSense error calling rs2_pipeline_start_with_config(pipe:0x1384d40, config:0x1910b90):
    hwmon command 0x10 failed. Error type:  (1).
cga@a1:~/opencv-rs2$ ./rs-multicam
^C
cga@a1:~/opencv-rs2$ ./rs-multicam
RealSense error calling rs2_pipeline_start_with_config(pipe:0x19f79d0, config:0x19fa2a0):
    hwmon command 0x10 failed. Error type:  (1).
cga@a1:~/opencv-rs2$ ./rs-multicam
RealSense error calling rs2_pipeline_start_with_config(pipe:0x1b65870, config:0x1b617a0):
    hwmon command 0x10 failed. Error type:  (1).
cga@a1:~/opencv-rs2$ ./rs-multicam
^C
cga@a1:~/opencv-rs2$ ./rs-multicam
^C
cga@a1:~/opencv-rs2$ ./rs-multicam
RealSense error calling rs2_pipeline_start_with_config(pipe:0xec80d0, config:0xecaa90):
    hwmon command 0x10 failed. Error type:  (1).

cga-cmu, 2019-06-25:
This code seems to work. I removed the OpenCV display stuff.

`int main(int argc, char * argv[]) try
{
  int i = 0;
  rs2:context ctx;
  int n_cameras;
  rs2::pipeline pipelines[MAX_N_CAMERAS];
  int loop_count = 0;
  int frame_counts[MAX_N_CAMERAS];
  
  // Create librealsense context for managing devices
  auto list = ctx.query_devices(); 
  n_cameras = list.size();
  if ( n_cameras == 0 ) 
    throw std::runtime_error("No device detected. Is it plugged in?");
  printf( "n_cameras: %d\n", n_cameras );
  // rs2::device dev = list.front();
  // cout << dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
  i = 0;
  for ( rs2::device dev : list )
    {
      cout << dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER) << '\n';
      devices[i] = dev;
      rs2::config cfg;
      cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
      cfg.enable_stream( RS2_STREAM_DEPTH, WIDTH_DEPTH, HEIGHT_DEPTH,
                         RS2_FORMAT_Z16, 6 );
      cfg.enable_stream( RS2_STREAM_COLOR, WIDTH_COLOR, HEIGHT_COLOR,
                         RS2_FORMAT_BGR8, 6 );
      cfg.enable_stream(RS2_STREAM_INFRARED, 1);
      cfg.enable_stream(RS2_STREAM_INFRARED, 2);
      configs[i] = cfg;
      rs2::pipeline pipe(ctx);
      pipelines[i] = pipe;
      // pipe.start( cfg );
      rs2::pipeline_profile profile = pipe.start( cfg );
      profiles[i] = profile;
      
      // generate window names and Opencv windows
      // OpenCV code goes here

      frame_counts[ i ] = 0;

      i++;
    }

  int last_frame_count = 0;

  while ( 1 )
    {
      // OpenCV waitKey(1) goes here

      ++loop_count;
      if ( frame_counts[0] % 10 == 1 && last_frame_count != frame_counts[0] )
        {
          printf( "%d: ", loop_count );
          for( int camera = 0; camera < n_cameras; camera++ )
            {
              printf( " %d", frame_counts[ camera ] );
            }
          printf( "\n" );
          last_frame_count = frame_counts[0];
        }
      for( int camera = 0; camera < n_cameras; camera++ )
        {
          rs2::frameset frames;
          // frames = pipelines[ camera ].wait_for_frames(); 
          if ( !(pipelines[ camera ].poll_for_frames( &frames )) )
            continue;

          // Process frames here

          ++(frame_counts[camera]);
        }
    }

  return EXIT_SUCCESS;
 }
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
`

cga-cmu, 2019-06-25:
Correction: above code sometimes gets error, but less than rs-multicam demo code.

cga@nuc0:~/opencv-rs2$ ./multi-all
n_cameras: 4
827112072509
827312071410
752112070756
827312071794
RealSense error calling rs2_pipeline_start_with_config(pipe:0x5556d6994c80, config:0x5556d698d8a0):
    hwmon command 0x10 failed. Error type:  (1).

cga@nuc0:~/opencv-rs2$ ./multi-all
n_cameras: 4
827112072509
827312071410
752112070756
827312071794
RealSense error calling rs2_pipeline_start_with_config(pipe:0x55a180592e10, config:0x55a180591560):
    hwmon command 0x30 failed. Error type:  (1).

kroq-gar78, 2019-06-25:
Here I create a `rs2::syncer` object to synchronize all sensors across all cameras. If you wanted to emulate the original `rs-multicam`'s use of `pipeline` objects, you could create a `syncer` for each _camera_, and put those into a `vector<syncer>` instead of `vector<pipeline>`.

```c++
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2015-2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp>     // Include RealSense Cross Platform API
#include "example.hpp"              // Include short list of convenience functions for rendering

#include <map>
#include <vector>

int main(int argc, char * argv[]) try
{
    // Create a simple OpenGL window for rendering:
    window app(1280, 960, "CPP Multi-Camera Example");

    rs2::context                ctx;            // Create librealsense context for managing devices

    rs2::colorizer              colorizer;      // Utility class to convert depth data RGB colorspace

    std::vector<rs2::pipeline>  pipelines;
    rs2::syncer sync;

    std::vector<rs2::sensor> sensors;

    // Start a streaming pipe per each connected device
    for (auto&& dev : ctx.query_devices())
    {
        /*rs2::pipeline pipe(ctx);
        rs2::config cfg;
        cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
        pipe.start(cfg);
        pipelines.emplace_back(pipe);*/

        for (rs2::sensor sensor : dev.query_sensors()) {
            auto sensor_stream_profiles = sensor.get_stream_profiles();

            if(auto depth_sensor = sensor.as<rs2::depth_sensor>()) {
                //depth_sensor.open(depth_sensor.get_stream_profiles()[68]); // 1280 x 720 30Hz
                depth_sensor.open(sensor_stream_profiles[77]); // 640 x 480 60Hz
                depth_sensor.start(sync);
                sensors.push_back(sensor);
            } else if (sensor_stream_profiles[0].stream_type() == RS2_STREAM_COLOR) { // "color sensor" doesn't have its own class
                sensor.open(sensor_stream_profiles[85]); // 640 x 480 60Hz RGB8

                sensor.start(sync);
                sensors.push_back(sensor);
            }
        }
    }

    // We'll keep track for the last frame of each stream available to make the presentation persistent
    std::map<int, rs2::frame> render_frames;

    // Main app loop
    while (app)
    {
        // Collect the new frames from all the connected devices
        std::vector<rs2::frame> new_frames;
        //for (auto &&pipe : pipelines)
        {
            rs2::frameset fs;
            if (sync.poll_for_frames(&fs))
            {
                for (const rs2::frame& f : fs)
                    new_frames.emplace_back(f);
            }
        }

        // Convert the newly-arrived frames to render-firendly format
        for (const auto& frame : new_frames)
        {
            render_frames[frame.get_profile().unique_id()] = colorizer.process(frame);
        }

        // Present all the collected frames with openGl mosaic
        app.show(render_frames);
    }

    for(auto sensor : sensors) {
        sensor.stop();
        sensor.close();
    }

    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception & e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
```

derens99, 2019-06-28:
So the workaround is to use syncer instead of pipelines? How does this affect affect a hardware sync/ hardware trigger? I often find myself needing to unplug and replug these devices in order to solve the startup crashes also.

kroq-gar78, 2019-06-28:
It doesn't seem to affect hardware sync. `pipeline` uses `syncer` under the hood already I think, and HW sync is done by the cameras themselves (not in `librealsense`). See [this reply/thread](https://github.com/IntelRealSense/librealsense/issues/2637#issuecomment-434203140) for some details on how the sync exactly works. For the D435, inter-camera HW sync is done between depth sensors. In that same thread, dorodnic mentions that frame capture times are handled differently between the RGB and depth sensors, and that there is some machinery in `syncer` to handle this.

Do you need to unplug/replug the cameras even for the example I posted above? I never did try doing that with the original example.

vhanded, 2019-08-16:
I am having the same problem, and my current solution is to unplug, try again, unplug, and try again.

RealSenseCustomerSupport, 2019-09-23:
----------------------------------------------

@kroq-gar7 @vhanded Sorry for late response! Is the issue resolved now? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-09-29:
----------------------------------------------

@kroq-gar7 @vhanded Do you still need support on this? Thanks!

cga-cmu, 2019-09-29:
> EXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD73PEVQ.gif]

I have worked around it.
Thanks,
Chris

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

@cga-cmu

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

@cga-cmu Thanks for update! Close the issue accordingly.

closed 2019-10-12 by RealSenseCustomerSupport

lsorgi, 2021-03-12:
I had a similar issue..... it went away just changing the order of device creation:

using D435 and L515. 
I moved the initialization of the L515 pipeline before the D435 and the exceptions went away