realsenseai/librealsense #4163: T265 and D415 doesn't work at the same time

Shagai, 2019-06-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 and D415  | 
| Firmware Version                   | 0.0.18.5715 / 05.11.04.00 | 
| Operating System & Version |   Debian 9  | 
| Kernel Version (Linux Only)    |  4.9.0-9-amd64                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.22                          | 
| Language                            |  C++                           | 
| Segment			|  Robotics                          | 

### Issue Description

I created this small program to test the cameras:

```
#include <librealsense2/rs.hpp>     // Include RealSense Cross Platform API

#include <vector>
#include <string>

int main(int argc, char * argv[])
{
    std::string serialNumber = argv[1];
    rs2::context                ctx;            // Create librealsense context for managing devices

    // Start a streaming pipe per each connected device
    rs2::pipeline pipe(ctx);
    rs2::config cfg;
    cfg.enable_device(serialNumber);
    pipe.start(cfg);

    // Main loop
    while (true)
    {
        rs2::frameset fs;
        pipe.poll_for_frames(&fs);
    }

    return EXIT_SUCCESS;
}
```

I execute the program for both cameras in two different terminals at the same time:

`$ ./example serialNumberOfD15`
`$ ./example serialNumberOfT265`

The first program to be run is ok, but when I launch the second one for the second camera I got this constantly:

> Device-A6A0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)

What can the problem be? I try in different computer and the same problem.

Thank you.

Samvid95, 2019-06-07:
I cannot exactly point out on what is the problem but I suspect that when for the second program starts running it is getting the `pipe` object of the first camera again and that might be the reason your program is crashing constantly. 

```
for (auto&& dev : ctx.query_devices())
{
        if(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER) == argv[1]){
            rs2::pipeline pipe(ctx);
            rs2::config cfg;
            cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
            pipe.start(cfg);
       }
}
```
Can you try it out this snippet?? So you can query all the device and start appropriate camera according to your need. 

Have you checked out the[ Multicam demo](https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam)? You can start and manage multiple camera from the single code check if that helps. 

Samvid

Shagai, 2019-06-07:
@Samvid95 

I tried your snippet and still I have the same problem.

In the project that I am working is mandatory to have each camera running in a different binary. That's why I don't use the same code of Multicam demo.

Thanks.

Samvid95, 2019-06-07:
Is it still the same error? 

I have another theory now, I think because the USB bus and it is interrupting the second camera to start the pipe. 

So what if after you start the first camera you put that thread to sleep for around 30 seconds before polling the frames and while it is asleep start the second program and let it setup. 

`std::this_thread::sleep_for (std::chrono::seconds(30));`

Can you try that out? It won't completely solve your problem but probably help us out in getting more clear idea on what is causing the problem exactly.

urock, 2019-06-10:
This issue is connected to [774](https://github.com/IntelRealSense/realsense-ros/issues/774)

ev-mp, 2019-06-10:
@Shagai hello,
In case of multi-cam scenario all the initially connected T265 devices are assigned to the very first allocated `context` object. This is a limitation of the current design.
The scenario of running multiple devices with T265 is available in `rs-multicam` demo as pointed out by @Samvid95 above.

Running T265 and D435 in separate processes may not work, and at the very least - you'll have to ensure invoking the T265 console before D435's. 
So if there are no specific requirements - we recommend to run all cameras in a single process.

We can think of additional solutions if you absolutely must run those in separate processes, but I'd suggest to review the recommended option first.

See #3734, #3434, #3437, #3892 for additional info.

orshefi, 2019-06-11:
Hello, I had encountered the same issue when trying to open two pipelines from different processes (mandatory requirement to our end product) using the t265 on both pipelines.
The method of loading the cameras is using the pipeline to open a specific serial number of a certain camera.

Issues:

1. If I want to open both t265 and d415 I need to initiate the t265 first or else I get 
`Device-A6A0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)`

2. if I want to open both t265 from different processes I get success with uploading the first of which but the second notify about a futex locking when tracing the application
`Device-A6A0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)`

I think I traced the issue back to libtm manager, when a libtm context is created I presume that it locks all vendor:product id that relevant to rst265 in order to use the notify function to alert on state change

ptaq666, 2019-06-11:
Hi @ev-mp , me and @Shagai are responsible for robotic teleoperations in hazardous environments. For us it is always more safe to run the code responsible for a single camera device in a separate binary, because if one of them crashes (for any reason, our bug, library issue), we still can execute a recovery scenario using another device. Of course as a workaround for now we can use all devices in a single binary, but maybe this design limitation should be reconsidered and improved in the future.

ev-mp, 2019-06-11:
@ptaq666, I totally agree that this constrain should dealt with on the architecture level.
We are aware of the limitations it imposes and eventually the way T265 connections work will be refactored, though I cannot provide estimates.

In the meanwhile running T265 before D415 should be an effective and immediate work-around. 
Ultimately, you can make the startup even more deterministic by building librealsense as static library twice - with and without TM2 Cmake flag enabled - and then build and link your executables against those two lib files accordingly. This will ensure that T265 would be picked up by only one of the applications.

orshefi, 2019-06-12:
@ev-mp How does it solves my need to load two different t265 from different processes?

ev-mp, 2019-06-13:
@orshefi hello,
Invoking multiple T265 devices in separate processes is currently not supported as stated. 
The provided workaround applies to the user with D415 and T265.

RealSenseCustomerSupport, 2020-01-27:
----------------------------------------------

Hi @Shagai

Were you able to get around your issue?  Have you attempted the latest version 2.32.1 which has updates around how T265 is accessed?

RoboticsIntelligence, 2020-02-17:
x@laptop:~/ws_realsense$ ros2 launch realsense_examples rs_demo_t265.launch.py 
[INFO] [launch]: All log files can be found below /home/x/.ros/log/2020-02-17-14-09-44-644737-robot-13318
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [13339]
[INFO] [realsense_node-2]: process started with pid [13340]
[realsense_node-2] terminate called after throwing an instance of 'rs2::error'
[realsense_node-2]   what():  No device connected
[ERROR] [realsense_node-2]: process has died [pid 13340, exit code -6, cmd '/home/x/ws_realsense/install/realsense_node/lib/realsense_node/realsense_node __ns:=/'].

Using latest version (2.32.1) with https://github.com/intel/ros2_intel_realsense/tree/refactor
Ubuntu 18.04  5.3.0-28-generic (Also tried earlier with Ubuntu 18.04, kernel 4.15 with same errors)

Is something broken in ros2_intel_realsense? (I have tried both refactor and dashing branches - same errors). T265 works fine with the realsense-viewer. T265 and Multiple D435i works fine in realsense-viewer.

I am basically trying to run the rs_t265_and_d400_launch.py

ahuizxc, 2020-02-22:
> Required Info
> Camera Model	T265 and D415
> Firmware Version	0.0.18.5715 / 05.11.04.00
> Operating System & Version	Debian 9
> Kernel Version (Linux Only)	4.9.0-9-amd64
> Platform	PC
> SDK Version	2.22 [...]

Hi I was the owner of ros2 realsense project and we have encountered this problem before when we tried to enable the t265 and d400 camera at the same time and already solved it. So here is our solution:
https://github.com/intel/ros2_intel_realsense/blob/e41d491a1cbdb0c4529a651adbc45693db99bd01/realsense_ros/src/rs_factory.cpp#L41
hope it will help you, thanks:)

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

Hi @Shagai, @RoboticsIntelligence

Were you able to get around your issue?  Have you attempted the information that ahuizxc has provided?

***If we don’t hear from you in 7 days, this issue will be closed.**

closed 2020-03-17 by RealSenseCustomerSupport