realsenseai/librealsense #4164: Intel realsense D415 stopped depth stream

sahilparekh, 2019-06-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Ubuntu16  | 
| Kernel Version (Linux Only)    | 4.15.0-29-generic                                      | 
| Platform                                 | PC  |
| SDK Version                            |  2.22                     | 
| Language                            |  python                          | 
| Segment			|  others                      | 

### Issue Description

Suddenly after 4 days the depth stream stopped working.

##### lsusb output

```
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 093a:7352 Pixart Imaging, Inc. 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 020: ID 8086:0ad3 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```
#### rs-enumerate-devices output
```
Device info: 
    Name                          : 	Intel RealSense D415
    Serial Number                 : 	902112060080
    Firmware Version              : 	05.11.01.100
    Recommended Firmware Version  : 	05.10.03.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/video4linux/video1
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD3
    Usb Type Descriptor           : 	2.1

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format   
    Infrared	  1280x720	@ 6Hz	   UYVY
    Infrared 1	  1280x720	@ 6Hz	   Y8
    Infrared	  1280x720	@ 6Hz	   BGRA8
    Infrared	  1280x720	@ 6Hz	   RGBA8
    Infrared	  1280x720	@ 6Hz	   BGR8
    Infrared	  1280x720	@ 6Hz	   RGB8
    Infrared	  640x480	@ 30Hz	   UYVY
    Infrared 1	  640x480	@ 30Hz	   Y8
    Infrared	  640x480	@ 30Hz	   BGRA8
    Infrared	  640x480	@ 30Hz	   RGBA8
    Infrared	  640x480	@ 30Hz	   BGR8
    Infrared	  640x480	@ 30Hz	   RGB8
    Infrared	  640x480	@ 15Hz	   UYVY
    Infrared 1	  640x480	@ 15Hz	   Y8
    Infrared	  640x480	@ 15Hz	   BGRA8
    Infrared	  640x480	@ 15Hz	   RGBA8
    Infrared	  640x480	@ 15Hz	   BGR8
    Infrared	  640x480	@ 15Hz	   RGB8
    Infrared	  640x480	@ 6Hz	   UYVY
    Infrared 1	  640x480	@ 6Hz	   Y8
    Infrared	  640x480	@ 6Hz	   BGRA8
    Infrared	  640x480	@ 6Hz	   RGBA8
    Infrared	  640x480	@ 6Hz	   BGR8
    Infrared	  640x480	@ 6Hz	   RGB8
    Infrared	  480x270	@ 60Hz	   UYVY
    Infrared 1	  480x270	@ 60Hz	   Y8
    Infrared	  480x270	@ 60Hz	   BGRA8
    Infrared	  480x270	@ 60Hz	   RGBA8
    Infrared	  480x270	@ 60Hz	   BGR8
    Infrared	  480x270	@ 60Hz	   RGB8
    Infrared	  480x270	@ 30Hz	   UYVY
    Infrared 1	  480x270	@ 30Hz	   Y8
    Infrared	  480x270	@ 30Hz	   BGRA8
    Infrared	  480x270	@ 30Hz	   RGBA8
    Infrared	  480x270	@ 30Hz	   BGR8
    Infrared	  480x270	@ 30Hz	   RGB8
    Infrared	  480x270	@ 15Hz	   UYVY
    Infrared 1	  480x270	@ 15Hz	   Y8
    Infrared	  480x270	@ 15Hz	   BGRA8
    Infrared	  480x270	@ 15Hz	   RGBA8
    Infrared	  480x270	@ 15Hz	   BGR8
    Infrared	  480x270	@ 15Hz	   RGB8
    Infrared	  480x270	@ 6Hz	   UYVY
    Infrared 1	  480x270	@ 6Hz	   Y8
    Infrared	  480x270	@ 6Hz	   BGRA8
    Infrared	  480x270	@ 6Hz	   RGBA8
    Infrared	  480x270	@ 6Hz	   BGR8
    Infrared	  480x270	@ 6Hz	   RGB8
    Depth	  1280x720	@ 6Hz	   Z16
    Depth	  640x480	@ 30Hz	   Z16
    Depth	  640x480	@ 15Hz	   Z16
    Depth	  640x480	@ 6Hz	   Z16
    Depth	  480x270	@ 60Hz	   Z16
    Depth	  480x270	@ 30Hz	   Z16
    Depth	  480x270	@ 15Hz	   Z16
    Depth	  480x270	@ 6Hz	   Z16

Stream Profiles supported by RGB Camera
 Supported modes:
    stream       resolution      fps       format   
    Color	  1280x720	@ 6Hz	   RGB8
    Color	  1280x720	@ 6Hz	   Y16
    Color	  1280x720	@ 6Hz	   BGRA8
    Color	  1280x720	@ 6Hz	   RGBA8
    Color	  1280x720	@ 6Hz	   BGR8
    Color	  1280x720	@ 6Hz	   YUYV
    Color	  640x480	@ 30Hz	   RGB8
    Color	  640x480	@ 30Hz	   Y16
    Color	  640x480	@ 30Hz	   BGRA8
    Color	  640x480	@ 30Hz	   RGBA8
    Color	  640x480	@ 30Hz	   BGR8
    Color	  640x480	@ 30Hz	   YUYV
    Color	  640x480	@ 15Hz	   RGB8
    Color	  640x480	@ 15Hz	   Y16
    Color	  640x480	@ 15Hz	   BGRA8
    Color	  640x480	@ 15Hz	   RGBA8
    Color	  640x480	@ 15Hz	   BGR8
    Color	  640x480	@ 15Hz	   YUYV
    Color	  640x480	@ 6Hz	   RGB8
    Color	  640x480	@ 6Hz	   Y16
    Color	  640x480	@ 6Hz	   BGRA8
    Color	  640x480	@ 6Hz	   RGBA8
    Color	  640x480	@ 6Hz	   BGR8
    Color	  640x480	@ 6Hz	   YUYV
    Color	  424x240	@ 60Hz	   RGB8
    Color	  424x240	@ 60Hz	   Y16
    Color	  424x240	@ 60Hz	   BGRA8
    Color	  424x240	@ 60Hz	   RGBA8
    Color	  424x240	@ 60Hz	   BGR8
    Color	  424x240	@ 60Hz	   YUYV
    Color	  424x240	@ 30Hz	   RGB8
    Color	  424x240	@ 30Hz	   Y16
    Color	  424x240	@ 30Hz	   BGRA8
    Color	  424x240	@ 30Hz	   RGBA8
    Color	  424x240	@ 30Hz	   BGR8
    Color	  424x240	@ 30Hz	   YUYV
    Color	  424x240	@ 15Hz	   RGB8
    Color	  424x240	@ 15Hz	   Y16
    Color	  424x240	@ 15Hz	   BGRA8
    Color	  424x240	@ 15Hz	   RGBA8
    Color	  424x240	@ 15Hz	   BGR8
    Color	  424x240	@ 15Hz	   YUYV
    Color	  424x240	@ 6Hz	   RGB8
    Color	  424x240	@ 6Hz	   Y16
    Color	  424x240	@ 6Hz	   BGRA8
    Color	  424x240	@ 6Hz	   RGBA8
    Color	  424x240	@ 6Hz	   BGR8
    Color	  424x240	@ 6Hz	   YUYV
```

##### dmesg output

```
[16365.225696] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/input/input35
[16365.226346] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[16365.226712] uvcvideo: Unable to create debugfs 3-20 directory.
[16365.226883] uvcvideo 3-1:1.3: Entity type for entity Processing 7 was not initialized!
[16365.226888] uvcvideo 3-1:1.3: Entity type for entity Extension 8 was not initialized!
[16365.226892] uvcvideo 3-1:1.3: Entity type for entity Camera 6 was not initialized!
[16768.481328] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[16768.532676] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[16768.584055] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[16768.635455] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[16768.686804] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[16768.738177] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[16768.789681] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[16768.841211] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[16768.892550] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[16768.943966] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[16768.995427] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[16769.046817] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[16769.098204] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[16769.149550] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[16769.200947] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[16769.252450] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[16769.303798] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[16769.355335] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[16769.406796] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).

```

MartyG-RealSense, 2019-06-07:
Do you mean that you ran the camera stream continuously for 4 days, please.  If so, 4 days is the outer limits of what I would expect the camera to run for without stopping.  400 Series cameras can run indefinitely as long as their temperature remains within recommended limits.  If a problem develops in a long-run session then it is usually related to an issue in the computing hardware such as a USB or operating system glitch.

sahilparekh, 2019-06-10:
No the camera was being used for research use, so it was ON only for 3-4hrs a day.

MartyG-RealSense, 2019-06-10:
Do you know if the PC was rebooted at any time?  There have been some cases where the camera works fine when the computer is first booted, but stops working after the computer is rebooted.  It may be elated to the computer's BIOS hardware configuration settings.  

I believe that if this occurs, unplugging the camera and plugging it back in can fix the problem.  This is obviously awkward though if your computer is in an automated environment such as a factory, where you just want it to run autonomously without having to manually insert and remove cables.

sahilparekh, 2019-06-11:
The PC was never rebooted with camera plugged. Also i tried unplugging and re-plugging it many times but still no Depth stream. :(

RealSenseCustomerSupport, 2019-07-02:
----------------------------------------------

@sahilparekh How about IR stream? Could you please provide a snapshot when enable IR stream only? And did you try other USB ports or PCs to see if any issue also? Thanks!

RealSenseCustomerSupport, 2019-07-08:
----------------------------------------------

@sahilparekh Could you please update? Looking forward to your reply. Thanks!

sahilparekh, 2019-07-08:
@RealSenseCustomerSupport , Sorry for late reply.
For the above queries:
1) yes i had tried on different machines and also on different usb port on same machine but yet the depth stream is giving error
2) Attaching snapshot with only infrared channel on
[image]
[image]

RealSenseCustomerSupport, 2019-07-12:
----------------------------------------------

@sahilparekh Any operations such as update firmware for the camera before the issue happened or you just start-stop streaming? And noticed that you're using USB2 connection. Did you get chance to try  to connect the camera to USB3? Is it the same issue?

RealSenseCustomerSupport, 2019-07-26:
----------------------------------------------

@sahilparekh Sorry that we don't get response for weeks. Will close the ticket at this point. Please feel free to create new ticket if you still have question or issues. Thanks!

closed 2019-07-26 by RealSenseCustomerSupport