realsenseai/librealsense #4165: t265_stereo.py segmentation fault "Cannot move to target thread"

buzzer, 2019-06-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5715 | 
| Operating System & Version |    Linux (Xubuntu 16.04.)  | 
| Kernel Version (Linux Only)    |  4.15.0-50                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.22.0                          | 
| Language                            |  Python 3.5.2                          | 
| Segment			|  Robot                          | 

### Issue Description

Following the the librealsense [SDK Installation](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) and the setup from [t265_stereo.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py), the command
  `python3 t265_stereo.py` produces a segmentation fault:

> QObject::moveToThread: Current thread (0x2eaea00) is not the object's thread (0x3068ce0).
> Cannot move to target thread (0x2eaea00)
> 
> Segmentation fault

Installation of the SDK went fine and the realsense-viewer works properly. I am using OpenCV 4.1.0.

RealSenseCustomerSupport, 2019-07-25:
----------------------------------------------

Hi @buzzer,

I don't observe this issue on my x86 test system. Which platform do you use?

RealSenseCustomerSupport, 2019-08-22:
----------------------------------------------

Hi @buzzer,

Is the issue clarified? any update?

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

Hi @buzzer,

If no further question, will close this.

closed 2019-12-05 by RealSenseCustomerSupport