realsenseai/librealsense #4166: D435 only connecting as USB 2.1 after building LibRealSense 2

aburbanol, 2019-06-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Intel RealSense D435 | 
| Firmware Version                   | 05.11.04.00 | 
| Operating System & Version |   Linux Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.15.0-51-generic | 
| Platform                                 | PC |
| SDK Version                            |2.22.0                          | 
| Language                            |  C/++                        | 
| Segment			|  Robot                     | 

### Issue Description
The camera is only connecting as a 2.1 USB device and I can not use the higher resolution or both IR images. This happened after building and compiling the libraries from the source.

I installed my camera using the packages and it was working perfectly in USB 3 with all the resolutions. Then I uninstall all the packages and build the library RealSense like [this](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md). During the installation, I got the problem in step 4 _(./scripts/patch-realsense-ubuntu-lts.sh)_ of the videobuf2_core being used. I solve it using sudo modprobe videobuf2_core as I check in another Issue. After that, I successfully compiled and install the libRealSense and the examples. 

I reconnect the USBC cable in both senses and reinstalled the drivers (./scripts/patch-realsense-ubuntu-lts.sh)  like proposed in #1181  #1391 and  #1148

here the asked info in the other issues:
lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 006: ID 0483:374b STMicroelectronics ST-LINK/V2.1 (Nucleo-F103RB)
Bus 001 Device 004: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 003: ID 8087:0a2a Intel Corp. 
Bus 001 Device 002: ID 04f2:b5b0 Chicony Electronics Co., Ltd 
Bus 001 Device 009: ID 8086:0b07 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 2: Dev 2, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 2: Dev 2, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 3: Dev 3, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 3: Dev 3, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 4: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 6, If 2, Class=Communications, Driver=cdc_acm, 12M
        |__ Port 1: Dev 6, If 0, Class=Vendor Specific Class, Driver=, 12M
        |__ Port 1: Dev 6, If 3, Class=CDC Data, Driver=cdc_acm, 12M
        |__ Port 1: Dev 6, If 1, Class=Mass Storage, Driver=usb-storage, 12M
        |__ Port 4: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 7, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 7, If 2, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 11: Dev 9, If 4, Class=Video, Driver=uvcvideo, 480M
    |__ Port 11: Dev 9, If 2, Class=Video, Driver=uvcvideo, 480M
    |__ Port 11: Dev 9, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 11: Dev 9, If 3, Class=Video, Driver=uvcvideo, 480M
    |__ Port 11: Dev 9, If 1, Class=Video, Driver=uvcvideo, 480M

rs-enumerate-devices
Device info: 
    Name                          : 	Intel RealSense D435
    Serial Number                 : 	827312072047
    Firmware Version              : 	05.11.04.00
    Recommended Firmware Version  : 	05.10.03.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.0/video4linux/video1
    Debug Op Code                 : 	15
    Advanced Mode                 : 	NO
    Product Id                    : 	0B07
    Usb Type Descriptor           : 	2.1

ev-mp, 2019-06-10:
@aburbanol hello,
The enumeration process and the patches kernel modules are not related and cannot affect each other.

The issue that you describe with camera being recognized as USB2.1 commonly occurs due to cables/connectors and can be often mitigated by re-plugging the camera or switching USB port. In case the USB2.1 is persistent then I'd suggest to replace the cable and/or switch to an external self-powered USB3 hub.

Note that USB3 type-C cables are less-robust and may require specific handling:
- Some Type-C <=> Type-C cables can be used for USB3 only in one direction.
- Type-C cable must be plugged-in into device/host fast and firmly. Slow connection may result in device being enumerated as USB2 by design.

aburbanol, 2019-06-11:
Hi @ev-mp, 
Following your instructions, I used 2 different self-powered USB3 hubs connected to 4 different port USB (in 2 different locations of the main board) with 3 different cables USBC, but still not able to connect the camera in a USB3 connection. I always manage to see the camera, but never USB3. 

In my understanding, this script (./scripts/patch-realsense-ubuntu-lts.sh) is the one we use to reinstall the drivers? I don't know what else can I do to be able to connect the camera un USB 3.

ev-mp, 2019-06-11:
@aburbanol hello, I'd suggest to cross-validate the HW setup to isolate the issue:
- Connect and verify USB3 enumeration for the camera on a separate Linux or Windows PC.
- Check that non-Realsense device can be properly enumerated as USB3 when connected to the same ports on the original host PC.

aburbanol, 2019-06-12:
Hi @ev-mp,

1. I connected to my W10 PC and it is identified as a USB 3.0 device (using the original cable).
2. I connected an Intel R200 and it was recognized as USB 3.0

The weirdest thing is that now the D435 is being recognized as USB 3.2. So I don't know if it was the fact of connecting another USB 3 camera or to connect the D435 to windows PC make it work. I restarted my Linux PC 3 times to be sure that it will work in the future.

 I don't know if you are interested to gather some data for you guys to check, but for the moment the issue is resolved. thanks for your help.