realsenseai/librealsense #4167: How to use playback function in python?

filipematosinov, 2019-06-07:
I am having trouble trying to use the playback function to retrieve the frames from a bag in Python. Can someone give a simple example how to implement it in Python? I am interested in getting the color frames and depth frames (as from the normal streaming) but also to be able to pause and resume the playback  (to be able to process the frames and don't loose them)

MartyG-RealSense, 2019-06-07:
The SDK has a Python example program for reading bags called **read_bag_example** - would this be of use to you, please?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py

filipematosinov, 2019-06-14:
Thanks @MartyG-RealSense Marty for your answer but that's not I am looking for. In the C++ API there is a class called playback that I would like to use in Python. 
Using that example I don't have control when the frames come and I easily loose them while processing the previous ones.

lramati, 2019-06-16:
You can modify that example to grab the device out of the pipeline by replacing line 46 with:
```python
pipe_profile = pipeline.start(config)
playback = pipe_profile.get_device.as_playback()
```

lramati, 2019-06-16:
alternatively, the ```context.load_device(filename)``` function also exists in python, which you could use to load and configure reading the bag file manually

filipematosinov, 2019-06-17:
@Iramati thanks for yous answers, they are more than useful :) I am just wondering how does the context function would help me out. I have tried before but not sure what's going on (in terms of loosing frames)

lramati, 2019-06-17:
You might also want to open the bag with a rosbag inspection tool to make
sure that the frames weren't dropped in the recording process

lramati, 2019-06-17:
You would be able to make the device and use set_real_time(False) on it
before starting the streams, but that would still require you to configure
all the streams manually

RealSenseCustomerSupport, 2019-07-02:
----------------------------------------------

@filipematosinov Did you get through with Iramati's suggestion? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-07-11:
----------------------------------------------

@filipematosinov If no other questions about this ticket, will close it accordingly. Thanks!

closed 2019-07-11 by RealSenseCustomerSupport

filipematosinov, 2019-07-19:
I am having some trouble when closing the rs.pipeline. Basically when I do pip.stop() (where pip = rs.pipeline()) the script just stays there without dong nothing.

lramati, 2019-09-02:
This should be addressed in PR #4758