realsenseai/librealsense #4169: error in installing pyrealsense2 from source

rijo11811, 2019-06-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubantumate  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi3 B+  |
| SDK Version                            |  {  2.2 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I am trying to install Pyrealsense2 from source and got stuck due to errors. I am following the below link.:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python
 
Getting stuck in step 3. Execution of the CMAKE command results in error in line 16 and line 57 of the CmakeList file inside the python folder in wrappers. In order to perform step 3, I executed following command from python folder:

cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true

Line16-->
set_target_properties(pyrealsense2 PROPERTIES VERSION
    ${REALSENSE_VERSION_STRING} SOVERSION "${REALSENSE_VERSION_MAJOR}.${REALSENSE_VERSION_MINOR}")
error:
Incorrect number of parameters. 

Line57-->
if(${BACKEND} STREQUAL RS2_USE_V4L2_BACKEND)

error:
Unknown parameters

It seems that the error is as a result of Cmake not able to get the values of REALSENSE_VERSION_STRING and BACKEND variables. 

Enclosing $REALSENSE_VERSION_STRING and $BACKEND in "" resolves the error but I do not think this is the right way forward.

Any help would be great.

ev-mp, 2019-06-10:
@rijo11811  hello,
Building python wrapper from source [requires invoking the CMake from the topmost](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source) (i.e. Librealsense's root) directory.

rijo11811, 2019-06-11:
@ev-mp hello, 
 Your input resolved the issue.

closed 2019-06-11 by rijo11811