realsenseai/librealsense #4170: Issues with D435 use on Upboard

JoeD1557, 2019-06-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.04.00 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    | 4.15.0-51-generic                              | 
| Platform                                 | UpBoard  |
| SDK Version                            |  2.22.0                       | 
| Language                            |  C                         | 
| Segment			|  others                         | 

### Issue Description
Hello,

I am trying to use an Intel D435 to capture depth and RGB footage on an Upboard. I installed the SDK on the Upboard with no issues, and footage on the Realsense-Viewer works great, but I have been having difficulties using the camera with Gstreamer. Whenever I try to open a pipeline using Gstreamer on the Upboard (I have been using a basic pipe like 'v4l2src device="/dev/video2" ! videoconvert ! ximagesink') a display window opens but freezes immediately when attempting to communicate with the camera to negotiate caps; I don't think this is a caps issue, as I have passed the camera exact caps with the same result.
I initially thought it was an issue with the camera being connected on one of the USB 2.0 ports, but I have purchased an adapter for the USB 3.0 OTG port and connected the camera there and still experienced failures, albeit due to massive amounts of frame loss and large delays in creating the pipeline that usually end with the camera dropping connection with the board momentarily and then reconnecting.
I have been able to successfully create Gstreamer pipelines with this D435 on other, more powerful computers with USB A 3.0 ports, so I am wondering if the issue has to do with the Upboard's hardware capabilities or if it is due to a different issue I wouldn't be aware of. Any advice would be greatly appreciated.

Thanks in advance!

JoeD1557, 2019-06-10:
Another note, connecting the D435 via the USB 3.0 OTG port causes a large amount of additional issues with Realsense-Viewer and the board as a whole, including crashes, hardware errors concerning the camera, and persistent disconnection from and reconnection to the camera.

RealSenseCustomerSupport, 2019-07-08:
----------------------------------------------

@JoeD1557 According to your description, the issue is not librealsense or RealSense camera related. Maybe you can try Upboard forum to see if anyone has clue about this.

JoeD1557, 2019-07-08:
Yeah, I ended up being able to resolve the issue by replacing the board with one that had USB 3.0 A ports on it.

closed 2019-07-08 by JoeD1557