realsenseai/librealsense #4171: rs-convert in cmd says it is working but no file is created

millsrandy, 2019-06-09:
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5715 | 
| Operating System & Version |  Windows 10 Enterprise 2016 LTSB v.1607 | 
| Kernel Version (Linux Only)    |                                      | 
| Platform                                 | LattePanda (PC) |
| SDK Version                            | 2.22.0 | 
| Language                            |  C# }                          | 
| Segment			| others                       | 

### Issue Description

I am using cmd to try to convert the .bag file I recorded in RS Viewer to .csv.  I put the following into the cmd prompt:

`rs-convert -v C:\Users\LattePanda\Desktop\test01 -i 20190609_142245.bag`

I have tried different file path options and nothing seems to be working.  Any thoughts?

RealSenseCustomerSupport, 2019-07-25:
----------------------------------------------

@millsrandy

Indeed, it looks something wrong there. Need to address.

OrientalSurprise, 2019-09-17:
Hi! Any update here? I'm running into the same issue with the same camera, however it's with all file types except generating .png files. The file I'm really trying to get are .ply files.
I get a log file in the same directory but nothing else

OrientalSurprise, 2019-09-18:
Seems to be specific to my T265 camera .bag output.
I tried the same with a D435 and it worked fine

flowb, 2019-10-12:
this also happens when coverting .bag files to .ply

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Hi @flowb, @OrientalSurprise,

Thanks for this issue update, will further check internally.

millsrandy, 2019-11-15:
@RealSenseCustomerSupport any update on rs-convert?  Trying to convert .bag file to csv to access the x, y, z coordinates and bring those into Cinema4D to assign to a camera object

millsrandy, 2019-11-15:
Now when I run rs-convert, the converter runs fine up to 99%, says "CSV converter" then gives "Segmentation fault (core dumped)."

The .log says "13/11 17:31:06, 588 WARNING [548002587072] (playback_device.cpp:198) Playback device does not provide a matcher

RealSenseCustomerSupport, 2020-01-27:
----------------------------------------------

Hi @millsrandy

Are you still seeing this issue with the latest version 2.32.1?

MartyG-RealSense, 2020-03-26:
A RealSense user on the Intel Support forum also had the matcher problem today (March 2020).  

https://support.intelrealsense.com/hc/en-us/community/posts/360042055394/comments/360011064213

So I have added the "Bug" label to this discussion so that it remains open and the RealSense team can track it.

r91andersson, 2020-05-20:
I'm facing the same issue. I'm using the latest SDK version (at this date) and a Realsense D435i camera. 
I've tried record the rosbag with command `rs-record -f /home/librealsense/test1.bag -t 10` and with realsense-viewer. Then I've tried to convert it to color frames using:` rs-convert -i /home/Documents/20200520_092425.bag -r /home/converted_ros_bag_realsense -c` 
However, when trying to convert the rosbag recorded with the rs-record, it converts all frames successfully, but the output is empty.
Then when trying to convert the rosbag recorded with realsense-viewer, the converter runs to 100% , and then Im getting the error: 
`[1]    11918 segmentation fault (core dumped)  rs-convert -i /home/Documents/20200520_092425.bag -r  -c`

MartyG-RealSense, 2020-05-20:
An alternative bag to image file extraction tool for RealSense 400 Series cameras is **bag2image**.

https://github.com/UnaNancyOwen/rs_bag2image

RealSenseSupport, 2020-09-10:
Thank you for highlighting the convert issue with using bag file associated with T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue for the T265.

millsrandy, 2020-09-10:
Appreciate you notifying me.  I eventually just gave up on the T265 as an option for the camera tracking project I was working on and have now switched over to using Vive Trackers.  Much better workflow.  Do you guys plan on offering an updated tracking device similar to the T265 in the future?

RealSenseSupport, 2020-09-11:
Unfortunately we cannot comment on products that haven't been released or upcoming products that may be in the pipeline.

closed 2020-09-15 by RealSenseSupport