realsenseai/librealsense #4172: issue about D435i IMU calibration

JoeyLeeNPU, 2019-06-10:
Now we want to use a D435i to test SVO2. 
One step is to calibrate IMU & monocular camera (both in D435i),  before this, some parameters should be input such as IMU's "gyroscope_noise_density", "accelerometer_noise_density", "gyrocope_random_walk", "accelerometer_random_walk".
In the BMI055 datasheet we can not find all parameters and they also don't exist in RealSense calibration results.
So how should I do?
Hope for your response, thanks!

dmirota, 2019-07-31:
Please refer to the [Kalibr](https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model) documentation for the explanation of these parameters.

ev-mp, 2019-09-19:
@JoeyLeeNPU , was the info referred by @dmirota  helpful ?
Please update

JoeyLeeNPU, 2019-09-19:
I'm sorry for response later, the problem has solved.

closed 2019-09-19 by JoeyLeeNPU