realsenseai/librealsense #4173: Document for D435 extension units

SorinAvramoni, 2019-06-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 }| 
| Firmware Version                   | (2.0) | 
| Operating System & Version |   Linux (Ubuntu 17)  | 
| Kernel Version (Linux Only)    |  (4.14.13)                                         | 
| Platform                                 | Intel Movidius MyriadX custom board  |
| SDK Version                            |   legacy / 2.0                          | 
| Language                            |  C                          | 
| Segment			|  Robot                           | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hi,

We are looking for a document that describes the extension unit for Real Sense Camera D435, something that would specify every Control Selector: a short description, supported requests (e.g. get min/max, get default, set current, get current), size and data type (e.g. Number, Integer, Float etc.).

Appreciate your help!
Sorin.

MartyG-RealSense, 2019-06-10:
The best documentation available for RealSense SDK 2.0 instructions is an unofficial site by RealSense user UnaNancyOwen.  It draws information from the official documentation and arranges it into an easily readable format that can be navigated with menus at the top of the page.

https://unanancyowen.github.io/librealsense2_apireference/

You will not be able to obtain documentation for every single function though, unfortunately.  The 'Advanced Mode' functions are not fully documented and there are no plans by Intel to do so.  This is because the complex ways in which Advanced Mode values interacts means that Intel handles them with machine learning algorithms instead.  Though you are free to do trial and error experimentation with Advanced Mode settings and see how they affect the image.

dorodnic, 2019-06-11:
Hi @SorinAvramoni 
Sounds like you are looking for documentation of firmware API. It is not publicly available. 
librealsense is capturing the de-facto API of each firmware version and converts it to one unified representation. You can query from librealsense supported controls, ranges and descriptions. 
For most controls this mapping is fairly straightforward, but some supported flows involve more logic.

SorinAvramoni, 2019-06-12:
Hi @dorodnic, somehow we found what we need, thanks a lot for your help!

closed 2019-06-12 by SorinAvramoni