realsenseai/librealsense #4174: T265 Inaccurate Scale

yx0123, 2019-06-10:
I am running T265 on Linux and found that in one of the tests, the tracked path is much smaller than the actual path (less than half the size). This has only happened once so far and I am not sure what caused the scale issue. Has anyone experienced this before?

Gnaarki, 2019-06-18:
I have the same issue. I tested the T265 outdoors and marked a 5m distance, then I walked the distance and saved the distance measured. My Result was that one time the distance was 5,04m but most time the distance was between 2,9m and 4m. I've done a second Test with a 3mx3m path with an angle of 90° between. The result was measured distances between 0,7m and max of 2,71m for the single sides. Most of the result were around 2m.

yx0123, 2019-06-18:
I tried holding the camera so that it points to the side instead of facing the direction I was walking, and the tracking seems to be better.

cimbar, 2019-06-20:
I am using T265 on an UAV and tried different sides of camera with various environments that have both much features and less features to detect. I also suffered from position scale error. In 12m flight path, T265 computes the distance between 7-10m. It is too much for this kind of short range.

RealSenseSupport, 2020-09-11:
Thank you for highlighting the issue of inaccurate scale while using T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

closed 2020-09-16 by RealSenseSupport