realsenseai/librealsense #4175: Measure the size of a small moving object

karimbernardet, 2019-06-10:
Hi

Using :
- D435 with firmware version 5.11.4
- Virtual machine with Ubuntu 18
- SDK 2.22
- Opencv

I am trying to track a small moving object (~ 2 cms) at ~ 2 meters from the camera (please see the black object surrounded in red). So I would like to have the size of the object and the distance from the camera.

I have tried rs-measure and the python script below
https://github.com/soarwing52/RealsensePython/blob/master/drafts/save-translate.py

I am using the following parms 
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 6)
config.enable_stream(rs.stream.depth,1280, 720 , rs.format.z16, 6)
But the results are not always reliable.
What could I try please to improve them ?

Cheers

Karim

[image]
[image]
[image]

MartyG-RealSense, 2019-06-10:
I would first try tracking an object that is colored something other than black or dark grey.  These colors can be harder for the depth camera to read, as it is a general physics principle that those colors absorb light.  An example is scanning a black cable.  It may appear on the image as looking like a black cable but is actually a cable-shaped black empty area.

karimbernardet, 2019-06-10:
Sorry I should have been more precise. I need to track hornets, bees for example

MartyG-RealSense, 2019-06-10:
I don't know if there is a precedent for purposely tracking a living object as small and fast as that, but I would think that it may be possible so long as you are just tracking their motion and not doing something as precise as measuring their height.

There was a user who was using D435 to track a mouse, which is a bit larger and slower than a bee but is an interesting demonstration of the possibilities.

https://github.com/IntelRealSense/librealsense/issues/1171#issuecomment-365468314

I would imagine that something as small as a bee might appear on the camera image as an 'artifact', like a quick-moving blur or smear that passes across the image.  Especially if you are using a D415, as it has a slow Rolling Shutter that can cause smears on the image if an object is traveling across its view quickly.  The D435 model will be better suited to this application, as it has a faster Global Shutter that is more suited to capturing fast-moving objects.

If very high quality tracking data is needed, a depth camera can be paired with the T265 Tracking Camera.

agrunnet, 2019-07-01:
I agree that tracking a bee is very challenging because it is very small, but it depends on how close you can be. Also you the challenge is compounded by the speed of the bees. Luckily we have started investigating this, and I can give you some preliminary data. This is particularly in the context of exploring adding a new high-speed mode that allows us to go to 400fps capture, but with a narrower vertical FOV.

Below is an indoor single shot capture of a small object  
[image]
We found that we really needed the object to be take up about 9 pixels in order to reliably track it. 

So let me go back to your original statement: "results are not always reliable". 
Reason could be
1. You are below the 9 pixel value - Try to move closer
2. The color is not aligned with the depth - check calibration and also show the LEFT monochrome image to see whether it is aligned to that. This will rule out camera extrinsic calibration because the left monochrome stream is perfectly aligned to the depth. 
3. You are operating at 1280x720 at 30fps. Try to use 848x480 at 90fps. You may be looking at color/depth frames captured at different times. 
4. Finally, if you find that 90fps is not fast enough, let us know. We are working on 4.4x faster and definitely like to help shore up the demand for it.

karimbernardet, 2019-07-07:
Hi
Thanks for your help
For the point 2, is there any procedure somewhere to do that ? (I am under ubuntu linux)
Thanks
Karim

agrunnet, 2019-07-07:
config->enable_stream(RS2_STREAM_DEPTH, 0, 848, 480, RS2_FORMAT_Z16, 90); \\ enable depth 
config->enable_stream(RS2_STREAM_INFRARED, 1, 848, 480, RS2_FORMAT_Y8, 90); \\ enable left monochrome

closed 2019-07-23 by RealSenseCustomerSupport