realsenseai/librealsense #4177: RealSense error calling

TianhangTang, 2019-06-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10) | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.19. 1}                          |    

### Issue Description
Hello, I am using the real-sense for testing, however, after I updated the camera device from 5.10.3 to 5.11.4, an error occurred. When I opened the 'Intel RealSense Viewer', I can not find the RGB options, and the software pops up a window which says RealSense error calling, here is the information of the error window:

Backend in rs2_get_option_range(options:000001E9FF3A24A0, option:Enable Auto White Balance, min:-nan, max:6.85235e-43, step:5.60519e-45, def:3.43702e+34):
ks_control->KsProperty( reinterpret_cast<PKSPROPERTY>(&node), sizeof(node), &description, sizeof(KSPROPERTY_DESCRIPTION), &bytes_received) returned: HResult 0x80070490: "~
0C ?"

I wonder whether it is a bug of the latest device version or there is something wrong with my device. By the way, I used another realsense device to test, it could work normally, Could you give some suggestions? Should I downgrade my realsense back to version 5.10.3?

ev-mp, 2019-06-10:
@TianhangTang , can you provide the output for `rs-enumerate-devices -c` for the two malfunctioning devices ?

RealSenseCustomerSupport, 2019-06-28:
----------------------------------------------

@TianhangTang This seems to be the known issue which will be fixed by later firmware version. Currently please recover it by below procedure.

- Use DFU tool to roll back firmware to previous firmware ( in you case it should be v5.10.3)
- DO NOT disconnect USB3 cable right after firmware update finishes - just leave it there for ~20 seconds
- After that it is safe to unplug the camera if you want.

TianhangTang, 2019-06-28:
@ev-mp Thank you for your reply, and sorry for not answering you these days because these devices do not belong to me, so I have to turn them back to company and have no chance to do the test. Anyway, they do work fine after I downgraded their firmware back to former version. Thanks again.

TianhangTang, 2019-06-28:
@RealSenseCustomerSupport Thank you for you answer and suggestions, these suggestions is helpful to me, looking forward to your later firmware version.

closed 2019-07-08 by TianhangTang