realsenseai/librealsense #4178: How set realsense RGB camera resolution with opencv?

TianhangTang, 2019-06-10:
Hello, I am trying to using opencv to get the realsense RGB frame, but the default resolution is 640*480. I treat the RGB camera as a normal camera, and use the cv::VideoCapture to control it, set the frame width and height to 1920*1080. But the result seems to be the resized result of the original frame 640*480. Does it mean the maximum resolution is 640*480 if I using other tools to get the camera frames? Do I have to use the tools provided in realsense SDK if I want to get the high resolution 1920*1080?

RealSenseCustomerSupport, 2019-07-01:
----------------------------------------------

@TianhangTang Considering the issue that only 640x480 resolution can be got by OpenCv, I think it should be OpenCv related. Here are two links for your reference. Please check to see if helps.
https://stackoverflow.com/questions/16390810/opencv-cant-set-resolution-of-video-capture
https://stackoverflow.com/questions/14287/increasing-camera-capture-resolution-in-opencv

RealSenseCustomerSupport, 2019-07-08:
----------------------------------------------

@TianhangTang If no other questions, will close this ticket accordingly. Thanks!

TianhangTang, 2019-07-08:
@RealSenseCustomerSupport I am not familiar with the github and sorry for forgetting to close the issue, Thank you for you kindly help.

closed 2019-07-08 by TianhangTang