realsenseai/librealsense #4179: How to get FOV from the pyrealsense2 object

itachi134999, 2019-06-10:
Hello,

I need the FOV values (horizontal, vertical and diagonal) of RGB color for real sense 415

MartyG-RealSense, 2019-06-10:
I recalled a past case where a Python user wanted to get the field of view.  It was a long discussion that culminated at the link below, where the user said that they got the answer that they needed.

https://github.com/IntelRealSense/librealsense/issues/1904?language=en_US#issuecomment-398434434

My understanding from the reading of the script in that link is that they get the color intrinsics and then print out its **ppx** and **ppy** values

[image]

itachi134999, 2019-06-11:
I got the way to get to fov horizontal and vertical but i am not getting diagonal data value:
FOV value [69.29666900634766, 42.46598434448242]

MartyG-RealSense, 2019-06-11:
I'm not sure that there is a function to get diagonal FOV in the SDK.  In scripting, it is usually described just in terms of the X (horizontal) and Y (vertical).  Such as fx / fy and ppx / ppy.

Apparently diagonal FOV in general can be calculated as a square root:

**Da = sqrt(Ha*Ha + Va*Va)**

In more visual-friendly math, it should look like this, I believe:

[image]

itachi134999, 2019-06-11:
with that value i am getting diagonal FOV as 81.XXXX but expected value is 77.XXXX. Is this the correct formula.

MartyG-RealSense, 2019-06-11:
The data sheet for the 400 Series says that the 77 degree diagonal FOV for the color camera can be + or - 1 degree.  It adds that "due to mechanical tolerances of + / - 5%, Max and Min FOV values can vary from lens to lens and module to module by ~ + / - 3 degrees".

agrunnet, 2019-06-11:
The correct way to calculate the FOV is to use the intrinsics for focal length, fx and fy, and image height h and width w. However since fx=fy, we call that f.

HFOV = 2*atan[w/(2f)] 
VFOV= 2*atan[h/(2f)]
For diagonal you need to take Diag=sqrt(w*w+h*h)
DFOV=2*atan[D/2f]

So an example value for D435 at 848x480, might be f=424.74.

This would give 
HFOV= 2*atan[848/(2*427.74)] = 89.49deg
VFOV= 2*atan[480/(2*427.74)]= 58.59deg
DFOV=2*atan[974.4/(2*427.74)]=97.4deg

closed 2019-07-13 by RealSenseCustomerSupport